platformio/platformio-core #1000: Cannot run platformio/pio with parameters e.g. pio init throws python exceptionctx.args[0] == "upgrade", "--json-output" in ctx_args, IndexError: list index out of range

terxw, 2017-07-26:
### Configuration

**Operating system**: Linux 4.9.13-muqss-xanmod19 x86_64 GNU/Linux

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.4.0

### Description of problem

I cannot run platformio or pio command with most of the arguments

I can ran `platformio` , `pio` and `platformio --version`

I get: 

```Usage: pio [OPTIONS] COMMAND [ARGS]...

Options:
  --version          Show the version and exit.
  -f, --force        Force to accept any confirmation prompts.
  -c, --caller TEXT  Caller ID (service).
  -h, --help         Show this message and exit.

Commands:
  account   Manage PIO Account
  boards    Embedded Board Explorer
  ci        Continuous Integration
  debug     PIO Unified Debugger
  device    Monitor device or list existing
  init      Initialize PlatformIO project or update existing
  lib       Library Manager
  platform  Platform Manager
  remote    PIO Remote
  run       Process project environments
  settings  Manage PlatformIO settings
  test      Local Unit Testing
  update    Update installed platforms, packages and libraries
  upgrade   Upgrade PlatformIO to the latest version
```

But if I try command with parameters for example: `pio init`

I get:

```
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 104, in main
    cli(None, None, None)
  File "/home/kangus/.local/lib/python2.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/kangus/.local/lib/python2.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/kangus/.local/lib/python2.7/site-packages/click/core.py", line 1063, in invoke
    Command.invoke(self, ctx)
  File "/home/kangus/.local/lib/python2.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/kangus/.local/lib/python2.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/kangus/.local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 74, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "/usr/local/lib/python2.7/dist-packages/platformio/maintenance.py", line 67, in on_platformio_start
    if not in_silence(ctx):
  File "/usr/local/lib/python2.7/dist-packages/platformio/maintenance.py", line 42, in in_silence
    ctx.args[0] == "upgrade", "--json-output" in ctx_args,
IndexError: list index out of range

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
### What I tried

I tried deleting ~/.platformio folder, reinstalling platformio, to no avail.

ivankravets, 2017-07-27:
Thanks!

Could you re-test with http://docs.platformio.org/en/latest/installation.html#development-version ?

closed 2017-07-27 by ivankravets

terxw, 2017-07-28:
Thank you, its working now!

commit 9994ed8b (2017-07-27): Check context arguments before comparison // Resolve #1000