platformio/platformio-core #1004: c/c++ project index rebuild failed

ogghst, 2017-08-01:
What kind of issue is this?

- [X] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

### Configuration

**Operating system**: Windows 7

**PlatformIO Version** (`platformio --version`): 3.4.1b1

### Description of problem

suddendly, after the inclusion of esp8266 board, a c/c++ error appears:

Error: Traceback (most recent call last):
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\__main__.py", line 104, in main
    cli(None, None, None)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 103, in cli
    pg.generate()
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 121, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 127, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "C:\Users\muratori.n\.platformio\penv\Scripts\bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "C:\Users\muratori.n\.platformio\penv\Scripts\bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "C:\Users\muratori.n\.platformio\penv\Scripts\bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 2, in <module>
AttributeError: 'NoneType' object has no attribute 'replace'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-08-01:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

ogghst, 2017-08-02:
C:\Users\muratori.n>echo %PATH%
c:\windows;C:\windows\system32;C:\ProgramData\Oracle\Java\javapath;C:\Program Fi
les\Git\cmd;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\LLVM\bin;C:\Pro
gram Files (x86)\Common Files\Acronis\SnapAPI\;C:\Program Files (x86)\Google\Chr
ome\Application;c:\windows;C:\windows\system32;C:\ProgramData\Oracle\Java\javapa
th;C:\Program Files\Git\cmd;C:\Program Files (x86)\Skype\Phone\;c:\windows;C:\wi
ndows\system32;C:\ProgramData\Oracle\Java\javapath;C:\Program Files\Git\cmd;C:\P
rogram Files (x86)\Skype\Phone\;C:\Program Files (x86)\Nmap;C:\Program Files (x8
6)\Nmap;C:\Users\muratori.n\AppData\Local\atom\bin

----

C:\Users\muratori.n>echo %PYTHONPATH%
%PYTHONPATH%

---
python, pip, pip list not recognized as a command

thanks!

ivankravets, 2017-08-02:
Please open PlatformIO IDE for Atom, open the problematic project, then PIO IDE Terminal and type
```
pio init --ide atom
```

Please share here all output.

ogghst, 2017-08-02:
PS E:\Development\mysensors_sanmartino> pio init --ide atom

The current working directory E:\Development\mysensors_sanmartino will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in E:\Development\mysensors_sanmartino
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries
Error: Traceback (most recent call last):
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\__main__.py", line 104, in main
    cli(None, None, None)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)

  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 103, in cli
    pg.generate()
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 121, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\users\muratori.n\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 127, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "C:\Users\muratori.n\.platformio\penv\Scripts\bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "C:\Users\muratori.n\.platformio\penv\Scripts\bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "C:\Users\muratori.n\.platformio\penv\Scripts\bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 2, in <module>
AttributeError: 'NoneType' object has no attribute 'replace'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-08-02:
Sorry, I forgot the main... You need to switch to debug version of PIO Core. Open PlatformIO IDE > Terminal and type:
```
pip install -U https://github.com/platformio/platformio-core/archive/feature/issue-1004-pg.zip
```

So, now `pio init --ide atom`

ogghst, 2017-08-02:
PS E:\Development\mysensors_sanmartino> pio init --ide atom
Please wait while upgrading PlatformIO...
Updating tool-pioplus                    @ 0.9.3          [0.9.4]
Uninstalling tool-pioplus @ 0.9.3:      [OK]
CorePackageManager: Installing tool-pioplus @ 0.9.4
Downloading...
Unpacking...
tool-pioplus @ 0.9.4 has been successfully installed!
PlatformIO has been successfully upgraded to 3.4.1rc1!

*********************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- support us with PlatformIO Plus > https://pioplus.com
*********************************************************************************************************************************************************************************************

The current working directory E:\Development\mysensors_sanmartino will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in E:\Development\mysensors_sanmartino
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries
13 {'env_pathsep': ';', 'project_name': 'mysensors_sanmartino', 'platformio_path': 'C:\\\\Users\\\\muratori.n\\\\.platformio\\\\penv\\\\Scripts\\\\platformio.exe', 'project_src_dir': 'E:\\De
velopment\\mysensors_sanmartino\\src', 'project_dir': 'E:\\Development\\mysensors_sanmartino', 'src_files': ['src\\node1\\node1_gateway_serial.cpp', 'src\\node10\\node10_gateway_ethernet.cpp
', 'src\\node11\\node11_hvac_ds18b20_camera.cpp', 'src\\node8\\node8_hvac_dht_sala.cpp', 'src\\node9\\node9_hvac_lm35_camera.cpp'], 'includes': [], 'user_home_dir': 'C:\\Users\\muratori.n',
'env_path': 'C:\\\\Users\\\\muratori.n\\\\.platformio\\\\penv\\\\Scripts;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\
\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Phone\\\\;C:\\\\Program Files\\\\LLVM\\\\bin;C:\\\\Program Files (x86)\\\\Common Files\\\\Acronis\\\\SnapAPI\\\\;C:\\\\Program Files (x86)\\\\Googl
e\\\\Chrome\\\\Application;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Pho
ne\\\\;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Phone\\\\;C:\\\\Program
 Files (x86)\\\\Nmap;C:\\\\Program Files (x86)\\\\Nmap;C:\\\\Users\\\\muratori.n\\\\AppData\\\\Local\\\\atom\\\\bin;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Jav
a\\\\javapath;C:\\\\Program Files\\\\Git\\\\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Phone\\\\;C:\\\\Program Files\\\\LLVM\\\\bin;C:\\\\Program Files (x86)\\\\Common Files\\\\Acronis\\\\Sna
pAPI\\\\;C:\\\\Program Files (x86)\\\\Google\\\\Chrome\\\\Application;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\\cm
d;C:\\\\Program Files (x86)\\\\Skype\\\\Phone\\\\;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\\cmd;C:\\\\Program File
s (x86)\\\\Skype\\\\Phone\\\\;C:\\\\Program Files (x86)\\\\Nmap;C:\\\\Program Files (x86)\\\\Nmap;C:\\\\Users\\\\muratori.n\\\\AppData\\\\Local\\\\atom\\\\bin', 'cxx_path': None, 'systype':
'windows_amd64', 'defines': []}

13 {'env_pathsep': ';', 'project_name': 'mysensors_sanmartino', 'platformio_path': 'C:\\\\Users\\\\muratori.n\\\\.platformio\\\\penv\\\\Scripts\\\\platformio.exe', 'project_src_dir': 'E:\\De
velopment\\mysensors_sanmartino\\src', 'project_dir': 'E:\\Development\\mysensors_sanmartino', 'src_files': ['src\\node1\\node1_gateway_serial.cpp', 'src\\node10\\node10_gateway_ethernet.cpp
', 'src\\node11\\node11_hvac_ds18b20_camera.cpp', 'src\\node8\\node8_hvac_dht_sala.cpp', 'src\\node9\\node9_hvac_lm35_camera.cpp'], 'includes': [], 'user_home_dir': 'C:\\Users\\muratori.n',
'env_path': 'C:\\\\Users\\\\muratori.n\\\\.platformio\\\\penv\\\\Scripts;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\
\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Phone\\\\;C:\\\\Program Files\\\\LLVM\\\\bin;C:\\\\Program Files (x86)\\\\Common Files\\\\Acronis\\\\SnapAPI\\\\;C:\\\\Program Files (x86)\\\\Googl
e\\\\Chrome\\\\Application;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Pho
ne\\\\;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Java\\\\javapath;C:\\\\Program Files\\\\Git\\\\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Phone\\\\;C:\\\\Program
 Files (x86)\\\\Nmap;C:\\\\Program Files (x86)\\\\Nmap;C:\\\\Users\\\\muratori.n\\\\AppData\\\\Local\\\\atom\\\\bin;c:\\\\windows;C:\\\\windows\\\\system32;C:\\\\ProgramData\\\\Oracle\\\\Jav
a\\\\javapath;C:\\\\Program Files\\\\Git\\\\cmd;C:\\\\Program Files (x86)\\\\Skype\\\\Phone\\\\;C:\\\\Program Files\\\\LLVM\\\\bin;C:\\\\Program Files (x86)\\\\Common Files\\\\Acronis\\\\Sna
[... cut ...]
AttributeError: 'NoneType' object has no attribute 'replace'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-08-02:
Thanks, useful dump!

> cxx_path': None

PIO can't find toolchains. Do you have antivirus in a system? It seems that it blocks PIO from downloading and installing toolchains to `C:\\Users\\muratori.n\.platformio\pakages`.

Now,
```
pio upgrade
pio platform list
```

ogghst, 2017-08-02:
thanks ivan,

i have an antivirus but i don't think it's blocking anything, at least not in my past 4 years.. i can access the folder and put files in explorer, i'll try to write some code to do it in atom.

this is the output of your commands:

PS E:\Development\mysensors_sanmartino> pio upgrade
Please wait while upgrading PlatformIO ...
PlatformIO has been successfully upgraded to 3.4.1
Release notes: http://docs.platformio.org/en/latest/history.html
PS E:\Development\mysensors_sanmartino> pio platform list
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 3.4.1!

*********************************************************************************************************************************************************************************************

If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- support us with PlatformIO Plus > https://pioplus.com
*********************************************************************************************************************************************************************************************

atmelavr ~ Atmel AVR
====================
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are bas
ed on the industrys most code-efficient architecture for C and assembly programming.

Home: http://platformio.org/platforms/atmelavr
Frameworks: arduino, simba
Packages: toolchain-atmelavr, framework-arduinoavr, framework-simba, tool-avrdude, tool-micronucleus
Version: 1.6.0

espressif8266 ~ Espressif 8266
==============================
Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things a
pplications.

Home: http://platformio.org/platforms/espressif8266
Frameworks: arduino, esp8266-rtos-sdk, simba
Packages: framework-esp8266-rtos-sdk, toolchain-xtensa, framework-simba, tool-esptool, tool-mkspiffs, tool-espotapy, framework-arduinoespressif8266, sdk-esp8266
Version: 1.3.1

ivankravets, 2017-08-02:
Please provide
```
pio platform show atmelavr
pio platform show espressif8266
```

Also, try to remove `C:\\Users\\muratori.n\.platformio\packages` folder and start build again.

ogghst, 2017-08-02:
atmelavr ~ Atmel AVR
====================
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are based on the industrys most code-efficient architecture for C and assembly programming.

Version: 1.6.0

Home: http://platformio.org/platforms/atmelavr

Repository: https://github.com/platformio/platform-atmelavr.git

Vendor: http://www.atmel.com/products/microcontrollers/avr/default.aspx

License: Apache-2.0

Frameworks: arduino, simba

Packages
--------

Package toolchain-atmelavr

--------------------------
Type: toolchain

Requirements: ~1.40902.0

Installed: Yes
Version: 1.40902.0

Original version: 4.9.2
Description: avr-gcc

Package framework-arduinoavr

----------------------------
Type: framework

Requirements: ~1.10619.0

Installed: Yes
Version: 1.10619.0

Original version: 1.6.19
Description: Arduino Wiring-based Framework (AVR Core, 1.6)

Package framework-simba

-----------------------
Type: framework

Requirements: >=7.0.0

Installed: No (optional)

Package tool-avrdude

--------------------
Type: uploader

Requirements: ~1.60300.0

Installed: Yes
Version: 1.60300.2

Original version: 6.3.0
Description: AVRDUDE

Package tool-micronucleus

-------------------------
[... 293 lines cut ...]
modwifi               ESP8266        80Mhz     2048kB  80kB   Olimex MOD-WIFI-ESP8266(-DEV)
thing                 ESP8266        80Mhz     512kB   80kB   SparkFun ESP8266 Thing
esp210                ESP8266        80Mhz     4096kB  80kB   SweetPea ESP-210

i've removed the packages folder and restarted atom (don't know what you mean for 'start build again'). this is the error message :

Uncaught SyntaxError: Unexpected token C in JSON at position 0

file:///C:/Users/muratori.n/AppData/Local/atom/app-1.18.0/resources/app/static/index.html:1
Hide Stack Trace
SyntaxError: Unexpected token C in JSON at position 0
    at Object.parse (native)
    at file:///C:/Users/muratori.n/.atom/packages/platformio-ide/lib/account/helpers.js:29:55
    at onExit (file:///C:/Users/muratori.n/.atom/packages/platformio-ide/lib/utils.js:92:7)
    at exit (file:///C:/Users/muratori.n/.atom/packages/platformio-ide/lib/utils.js:102:21)
    at triggerExitCallback (C:\Users\muratori.n\AppData\Local\atom\app-1.18.0\resources\app\src\buffered-process.js:338:17)
    at ChildProcess.<anonymous> (C:\Users\muratori.n\AppData\Local\atom\app-1.18.0\resources\app\src\buffered-process.js:368:17)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

the error was thrown from the platformio-ide package. this issue has already been reported:
https://github.com/platformio/platformio-atom-ide/issues/613

thanks

ivankravets, 2017-08-02:
Please navigate to project from PIO IDE Terminal and type `pio run`. Does it build a project?

ogghst, 2017-08-02:
PS E:\Development\mysensors_sanmartino> pio run
Can not remove temporary directory `E:\Development\mysensors_sanmartino\.pioenvs`. Please remove `.pioenvs` directory from the project manually to avoid build i
ssues

========================================================================== [SUMMARY] ==========================================================================
Environment pro8MHzatmega328_node1      [SKIP]
Environment pro8MHzatmega328_node8      [SKIP]
Environment pro8MHzatmega328_node9      [SKIP]
Environment nodemcu_node10              [SKIP]
Environment nodemcu_node11              [SKIP]
================================================================= [SUCCESS] Took 0.00 seconds =================================================================

should i remove that folder? do you have any experience in this folder access issue? 

if i run atom as administrator, the .pioenv warning disappear but c/c++ project index rebuild keeps going

ivankravets, 2017-08-02:
You should not use administrative permissions. Just remove `.pioenvs` and build project again without admin rights.

Also, would be good to share your `platformio.ini` here.

ogghst, 2017-08-02:
i tried admin right when you pointed out the file access issue..

however, removed .pioenvs, also packages and penv folder from the user directory, the issue still show.

this is my platformio.ini file. please note that i have a selective build strategy

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[platformio]
env_default = pro8MHzatmega328

[env:pro8MHzatmega328_node1]
platform = atmelavr
board = pro8MHzatmega328
framework = arduino
src_filter = +<*> +<node1/> -<node8/> -<node9/> -<node10/> -<node11/>
upload_port = COM7

[env:pro8MHzatmega328_node8]
platform = atmelavr
board = pro8MHzatmega328
framework = arduino
src_filter = +<*> +<node8/> -<node1/> -<node9/> -<node10/> -<node11/>
upload_port = COM7

[env:pro8MHzatmega328_node9]
platform = atmelavr
board = pro8MHzatmega328
framework = arduino
src_filter = +<*> +<node9/> -<node1/> -<node8/> -<node10/> -<node11/>
upload_port = COM12

[env:nodemcu_node10]
platform = espressif8266
board = nodemcu
framework = arduino
src_filter = +<*> +<node10/> -<node9/> -<node1/> -<node8/> -<node11/>
upload_port = COM5

[env:nodemcu_node11]
platform = atmelavr
board = pro8MHzatmega328
framework = arduino
src_filter = +<*> +<node11/> -<node9/> -<node1/> -<node8/> -<node10/>
upload_port = COM7

ivankravets, 2017-08-02:
I do not know how to help you. It seems that problem is linked with permissions/antivirus. PIO can't build a project, can't access or even install toolchains.

Try to remove `.pioenvs` directories, then whole `~/.platformio` folder,  then make clean uninstallation of PIO IDE for Atom ( http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide ).

After that, please install the latest Atom and our PIO IDE package.

DO NOT USE ADMIN RIGHTS when will install all from the scratch.

ivankravets, 2017-08-11:
Have you resolved this issue?

ogghst, 2017-08-12:
no, still happening randomly but all functionalities seems to be working..

closed 2017-08-12 by ivankravets