platformio/platformio-core #1006: Upload fails randomly

ChillarAnand, 2017-08-03:
- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:

```
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.3 LTS
Release:        16.04
Codename:       xenial
```

**PlatformIO Version** (`platformio --version`):

```
PlatformIO, version 3.4.1
```

### Description of problem
```
Auto-detected: /dev/ttyACM3
Uploading .pioenvs/megaatmega2560/firmware.hex
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer

avrdude done.  Thank you.

*** [upload] Error 1
```

#### Steps to Reproduce

Start a project.
Add some code.
Upload code.

### Actual Results
Upload fails

### Expected Results
Upload code 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
```

**Source file to reproduce issue:**
Any arduino code

ivankravets, 2017-08-03:
Have you tried to reconnect board and use usb cable with good quality?

ChillarAnand, 2017-08-04:
Yes, I am using a good quality cable. I guess it not related platformio. Thanks for the help.

closed 2017-08-04 by ChillarAnand

wisegeeksrules, 2018-04-10:
I'm having the same problem, trying to upload code onto a chinese "SparkFun Pro Micro 16MHz, 5V.

I have no problems using the Arduino plugin in Visual Studio Code, the issue is only with the PlatformIO IDE.