platformio/platformio-core #1009: CurieIMU method error on Arduino/Genuino 101

DaisukeUra, 2017-08-04:
### Configuration

**Operating system**: Win 7

**PlatformIO Version** (`platformio --version`): can't see because pio command not running. (but build, initialize runs correctly.

### Description of problem

I tried sample program of Arduino/Genuino 101 CurieIMU .
https://www.arduino.cc/en/Tutorial/Genuino101CurieIMUAccelerometer

It can build via arduino IDE.
But platformio can't find readAccelerometerScaled function and can't compile it.

#### Steps to Reproduce

1. Initialize directory by platformio on Atom or VSCODE for genuino 101 .
2. main.ino is place in src. 
3. copy and paste sample source to main.ino
4. Build it.

### Actual Results

C:/Users/ura/Documents/Arduino/platfom101/src/main.ino: In function 'void loop()':
C:/Users/ura/Documents/Arduino/platfom101/src/main.ino:63:14: error: 'class CurieIMUClass' has no member named 'readAccelerometerScaled'
*** [.pioenvs\genuino101\src\main.ino.o] Error 1

### Expected Results

Build complete.

### If problems with PlatformIO Build System: 

pio and platformio command not works on cmd or cygwin.
But, build and project initialize works correctly.

**The content of `platformio.ini`:**
```ini
[env:genuino101]
platform = intel_arc32
board = genuino101
framework = arduino
```

**Source file to reproduce issue:**
```cpp
/*
 * Copyright (c) 2016 Intel Corporation.  All rights reserved.
 * See the bottom of this file for the license terms.
 */

/*
   This sketch example demonstrates how the BMI160 on the
   Intel(R) Curie(TM) module can be used to read accelerometer data
*/

#include "CurieIMU.h"

void setup()
{
    Serial.begin(9600); // initialize Serial communication
    while (!Serial)
        ; // wait for the serial port to open

    // initialize device
    Serial.println("Initializing IMU device...");
    CurieIMU.begin();

    // Set the accelerometer range to 2G
    CurieIMU.setAccelerometerRange(2);
}

void loop()
{
    float ax, ay, az; //scaled accelerometer values

    // read accelerometer measurements from device, scaled to the configured range
    CurieIMU.readAccelerometerScaled(ax, ay, az);

    // display tab-separated accelerometer x/y/z values
    Serial.print("a:\t");
    Serial.print(ax);
    Serial.print("\t");
    Serial.print(ay);
    Serial.print("\t");
    Serial.print(az);
    Serial.println();
}

/*
   Copyright (c) 2016 Intel Corporation.  All rights reserved.

   This library is free software; you can redistribute it and/or
   modify it under the terms of the GNU Lesser General Public
   License as published by the Free Software Foundation; either
   version 2.1 of the License, or (at your option) any later version.

   This library is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
   Lesser General Public License for more details.

   You should have received a copy of the GNU Lesser General Public
   License along with this library; if not, write to the Free Software
   Foundation, Inc., 51 Franklin Street, Fifth Floor, Boston, MA  02110-1301  USA

*/
```

### Additional info

Best regards,
Daisuke

ivankravets, 2017-08-05:
Please move this issue to https://github.com/platformio/platform-intel_arc32/issues

closed 2017-08-05 by ivankravets

DaisukeUra, 2017-08-08:
Ok, now I submitted.
https://github.com/platformio/platform-intel_arc32/issues/5

thanks.