platformio/platformio-core #1010: Eclipse IDE: define __cpluscplus to build correct index

euphi, 2017-08-05:
With the default eclipse configuration created with `pio init --ide=eclipse` there is no symbol "__cplusplus" defined. Therefore the integrated eclipse indexer cannot handle C++11 standard lib symbols like `std::unique_ptr<T>`.

To solve this, it helps to define __cplusplus to a recent version, e.g.  "201402L".

platformio shall define this by default when initializing a project.

ivankravets, 2017-08-05:
Could you try http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags ?

euphi, 2017-08-06:
This does not help the eclipse indexer and it results in a warning from gcc:
`<command-line>:0:0: warning: "__cplusplus" redefined [enabled by default]`

Honestly, I have no clue how the eclipse indexer really works, but it seems that it uses the default gcc of the local system (and thus not the one installed as pio's platform) to build the index. For some reason I dont't know, it does not enable the latest C++ version by default. Thus, defining __cplusplus for the indexer tool "G++" helps.

This can be done by adding the follwing line to the `.cproject` file:

`<listOptionValue builtIn="false" value="__cplusplus=201402L"/>`

in the section of the symbols of the tool "G++":

```
<tool id="org.eclipse.cdt.build.core.settings.holder.54121539" name="GNU C++" superClass="org.eclipse.cdt.build.core.settings.holder">

[.... definition of include paths ...]

  <option id="org.eclipse.cdt.build.core.settings.holder.symbols.1198905600" name="Symbols" superClass="org.eclipse.cdt.build.core.settings.holder.symbols" valueType="definedSymbols">
    <listOptionValue builtIn="false" value="__cplusplus=201402L"/>
  [... definition of other symbols ...]
</option>
</tool>
```

See also euphi/ESP-LEDCtrl@287fd53c91472580eccb7e7aa6fcadbe4cb90bc4 for a commit I did on one of my projects to fix the indexer (be aware that this commit also contents some other cleanup of no longer used include paths etc.).

euphi, 2017-08-06:
See also https://stackoverflow.com/questions/17131744/eclipse-cdt-indexer-does-not-know-c11-containers
(thats where I found the solution).

ivankravets, 2018-02-08:
Could you reproduce this issue with the latest PIO Core 3.5.1? Please `pio upgrade` and re-init project.

JustACatRu, 2018-12-02:
Hello.

> Could you reproduce this issue with the latest PIO Core 3.5.1? Please `pio upgrade` and re-init project.

PlatformIO, version 3.6.2
Eclipse IDE for C/C++ Developers Version: 2018-09 (4.9.0) Build id: 20180917-1800

Same problem with "Symbol 'unique_ptr' could not be resolved" and other warnings.
Changing:
`...xtensa-lx106-elf-g++.exe ${FLAGS} -E -P -v -dD...`
to:
`...xtensa-lx106-elf-g++.exe ${FLAGS} -std=c++11 -E -P -v -dD...`
in project file "**/.settings/language.settings.xml**"solves the problem.
It is in "**provider**" parameter (2 times).

But when I use target "PlatformIO: Rebuild C/C++ Project Index", PlatformIO reverts my changes in "language.settings.xml" :(

As I understand, this file is formed from:
"platformio/ide/tpls/eclipse/.settings/language.settings.xml.tpl"
So, maybe you can just alter this .tpl for eclipse and add "-std=c++11" in it?

JustACatRu, 2018-12-03:
> See temporary solution here...

Thanks for quick reply, but I've tried this and other methods. It **does not work**.
I've recorded video to show this problem (your solution from 06:50 and my solution from 09:35): https://youtu.be/M2OzvkGHcSQ

Moreover adding "-std=c++11" to "build_flags" results in duplicating "-std=c++11" in build process (it is shown on video from 05:20).

I know, you are busy now. But It will be great, if you will merge my pull request or just add this "-std=c++11" to https://github.com/platformio/platformio-core/blob/develop/platformio/ide/tpls/eclipse/.settings/language.settings.xml.tpl when you'll have free time.

ivankravets, 2019-01-10:
Please re-test with the latest development version via `pio upgrade --dev`. Don't forget to re-init project.

closed 2019-01-10 by ivankravets

JustACatRu, 2019-01-11:
> Please re-test with the latest development version via `pio upgrade --dev`. Don't forget to re-init project.

I've re-tested, and there is another error now :(
`cxx_path.replace(user_home_dir, '')`
changed to:
`cc_path.replace(user_home_dir, '')`
for some reason.
And adding -std=c++11 not help now.

I've recorded video again for better explanation.
Please, check it: https://youtu.be/ah9Qod0p_I8

Thank you.

ivankravets, 2019-01-11:
Fixed in https://github.com/platformio/platformio-core/commit/286a53991c834120814e8f100e29fab7500bd8d5

Please repeat `pio upgrade --dev` and re-init project again. 

Does it work now?

JustACatRu, 2019-01-11:
> Does it work now?

Checked and...
Yep! Now the problem is solved! :)
PS: can provide video, if needed.

ivankravets, 2019-01-11:
Happy coding with PlatformIO! 😊

commit a092f87c (2019-01-10): Eclipse: Provide language standard to a project C/C++ indexer // Resolve #1010
commit 56cd55ba (2019-01-10): Eclipse: Provide language standard to a project C/C++ indexer // Resolve #1010