platformio/platformio-core #1011: Can't get rid of login nagscreen

telamon, 2017-08-09:
The shitty nagscreen  to login or create an account keeps popping up now everytime i start atom after I installed your plugin. I've kept pressing "skip" but it just returns again the next time i restart atom.

ivankravets, 2017-08-09:
Just navigate to PIO IDE Options and disable https://github.com/platformio/platformio-atom-ide/blob/develop/lib/config.js#L106

This login screen will be removed with the next update for IDE. We introduced new [PIO Home](https://community.platformio.org/t/platformio-home-2-0/2380/1) which will not require PIO Account.

closed 2017-08-09 by ivankravets

telamon, 2017-08-10:
Well to late.. platform-io is a buggy joke, once you're past the nagscreens it dosen't get better.
Exceptions flying everywhere, serial-port autodetection inconsistient breaking binary-flashing, overbloated features and project folders destroyed. 
Anyways I decided to uninstall platform-io seeing as the product is in no-where near production-quality, but
sadly it left my Atom in a horrible broken state. Has anyone of you ever even tried uninstalling pio???? 

Well.. thanks for nothing, what a complete waste of time.

ivankravets, 2017-08-12:
Thousand developers use PlatformIO each day. Can not imagine how they use this awful, buggy and catastrophic software...

ivankravets, 2017-08-12:
... and horrible, yes!

telamon, 2017-08-18:
Thank you for the recognition but yes, a week later I realize that not only where my expectations to high but I also chose the wrong approach, your atom-integration really is horrible but most notably atom just plainly does a bad job at doing anything without throwing exceptions, I should have chosen to use the "platform-io-core" cli instead. 

The most painful experience so far is caused by the fact that your homepage is pretty misleading, It took me a while to realize the concept of your "platforms" meaning you've either ported or wrapped upp custom build-chains in scons, which in itself isn't necessarily a bad idea, but in it's current state your force your user to relinquish all control over the actual build-chain as there is no easy way to customize your build-environment unless you guys have exposed some options. I've only tested platform io for the ESP32 and Arduino platform(you did an ok job for the Arduino, but your serial-port auto-detect is broken beyond recognition, I had to store serial settings in my platform.ini in order to be able to upload a sketch which in it self is really bad, (storing private settings in a git-revisioned file shouldn't be done.) And really arduino already has good CLI-support so that kindof makes platform-io completely redundant.  Also platform-io completley ignores the ini setting`lib_dir=~/Arduino/Libraries`.

As for the Espressif/ESP32 platform ... well there's a perfect example of bad integration right here in your issues-tracker, AFAIK there is no way to use the vendor provided `make menuconfig` tool to control your chip settings and there probably will never be deeming from the comments.

And third.. Platform-io downloads and indexes a brave repository of 3rd-party libraries. Intresting.. But ... Wait... There's no project-specific dependency management! Anyone remember the early 2000's and Netbeans/Eclipse plugin installation hell? Wait a moment there's more, there isn't any record of which platform version my project depends on either, so theoretically my project can never be run safely on anything other than my own computer given that I do not accidentally update any dependencies.

Also, where do all those gigabytes of toolchains and plugins and libraries i downloaded go? ( Now i know that it's spread out into hidden-folders in your home path, thanks for the heads up guys. )

So well if I consider Platform-io to be an highlevel abstractlayer-toolkit ala cordova - well then I'd say you're approaching beta status. Keep going, focus on the platform-io-core it's your greatest strength and your greatest weakness.

But as an complete IDE-solution, I guess in a world where the ultimate goal is to blink leds on hundreds of devices I admit it's true that thousand happy collaborating developers use PlatformIO each day.

ivankravets, 2017-08-18:
We have better integration with VSCode, see [PlatformIO IDE for VSCode](http://docs.platformio.org/en/latest/ide/vscode.html). New PIO Home, no required login, etc...

Could you try it out? Would be thankful for the feedback.

telamon, 2017-08-18:
I'm sorry I cannot help you with VSCode -  I truly fear for my health and well-being if I would be exposed to a Microsoft product on my local-systems.

But I can promise you to create some "a bit more civilized" issues regarding platformio-core in the future.

Ironically enough I find myself looking forward to future releases Platformio-Atom-IDE, but until then I'm going to stick to the core.

Peace

ivankravets, 2017-08-18:
>  your atom-integration really is horrible 

We are not experts in IDE, we don't work on own standalone IDE. We provide unified build system and debugger which could be integrated with the most popular IDEs. See the full list:
- http://docs.platformio.org/page/ide.html

> but most notably atom just plainly does a bad job at doing anything without throwing exceptions

Please file an issue here https://github.com/platformio/platformio-atom-ide/issues

>  but in it's current state your force your user to relinquish all control over the actual build-chain as there is no easy way to customize your build-environment unless you guys have exposed some options.

Very bad to hear that from "System Architect" and "Senior Developer". 

- http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

> but your serial-port auto-detect is broken beyond recognition

What is your board? Please provide `$ pio device list`.

> I had to store serial settings in my platform.ini in order to be able to upload a sketch which in it self is really bad, (storing private settings in a git-revisioned file shouldn't be done.)

* http://docs.platformio.org/en/latest/userguide/cmd_run.html#cmdoption-platformio-run-upload-port
* http://docs.platformio.org/en/latest/envvars.html#uploading
* http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

> has good CLI-support so that kindof makes platform-io completely redundant

A user decides which CLI to use. We don't force anyone.

> Also platform-io completley ignores the ini setting `lib_dir=~/Arduino/Libraries.`

* http://docs.platformio.org/en/latest/projectconf/section_env_library.html#projectconf-lib-extra-dirs

> make menuconfig tool to control your chip settings and there probably will never be deeming from the comments.

We are working together on that with Espressif team. Stay tuned with us https://twitter.com/PlatformIO_Org

> There's no project-specific dependency management! Anyone remember the early 2000's and Netbeans/Eclipse plugin installation hell?

OMG! Catastrophic! PlatformIO must die! 🔥 

- http://docs.platformio.org/en/latest/librarymanager/index.html

> Wait a moment there's more, there isn't any record of which platform version my project depends on either

See excellent example by @xoseperez  https://bitbucket.org/xoseperez/espurna/src/e6e9f9d6d4bd16005446e08b98d47c6cd254ad16/code/platformio.ini?at=master&fileviewer=file-view-default

Or, Marlin firmware:
- https://github.com/MarlinFirmware/Marlin/blob/2.0.x/platformio.ini

> Also, where do all those gigabytes of toolchains and plugins and libraries i downloaded go?

http://platformio.org => Bintray link.

===== 

Tony, I really don't like your tone. PlatformIO is open source project, Apache 2.0 license. A lot of people helped us to make this project better and better. Saying that all is bad and we should die - that is the worst that I've ever heard. 

Indeed, we, PlatformIO, nothing owe you. See
- https://github.com/platformio/platformio-core/blob/develop/LICENSE

> 7. Disclaimer of Warranty. Unless required by applicable law or
agreed to in writing, Licensor provides the Work (and each
Contributor provides its Contributions) on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or
implied, including, without limitation, any warranties or conditions
of TITLE, NON-INFRINGEMENT, MERCHANTABILITY, or FITNESS FOR A
PARTICULAR PURPOSE. You are solely responsible for determining the
appropriateness of using or redistributing the Work and assume any
risks associated with Your exercise of permissions under this License.