platformio/platformio-core #1017: Unable to install platformio on Rasberry Pi Raspian Stretch

blackketter, 2017-08-17:
What kind of issue is this?

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Raspian Stretch 

**PlatformIO Version** (`platformio --version`):
Current release, 3.4.1

### Description of problem
Attempting to install Platformio on fresh Raspian Stretch release.

#### Steps to Reproduce

1.  succeeds: 
`pip install -U platformio `

2. fails: 
`sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"`

### Actual Results
```
pi@raspberrypi:~ $ sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"

==> Installing Python Package Manager ...
Requirement already up-to-date: pip in /usr/lib/python2.7/dist-packages

[SUCCESS]

==> Installing PlatformIO and dependencies ...

/usr/bin/python: can't open file 'pip': [Errno 2] No such file or directory

[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio-core/issues >
```

ivankravets, 2017-08-17:
> sudo python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"

Where did you find this link?

This is really a bug in our installer. We fixed in develop version. Try to replace `master` with `develop` in that URL.

ivankravets, 2017-08-17:
Indeed, issue was fixed in https://github.com/platformio/platformio-core/commit/9b2e7bf927695f5b8d9e131613a43732035a6470

I see a few links on Installation page which lead to "master". I'll replace them to "develop".

Thanks a lot for the report!

closed 2017-08-17 by ivankravets

blackketter, 2017-08-17:
I believe I grabbed that link from the http://docs.platformio.org/en/latest/installation.html page, but I can't be sure now, since that appears to have "develop" in it now.  Did you change the docs?

I think I just hit DuckDuckGo with "install platformio cli"

I'm further confused by the title of that page being "<title>Installation &mdash; PlatformIO 3.5.0a4 documentation</title>".  :)

ivankravets, 2017-08-18:
Yes, yes, I fixed links in docs :) Thanks!

commit 44909f9c (2017-08-17): Switch to "develop" version of PIO Core Installer while it isn't merged to "master" // Resolve #1017