platformio/platformio-core #1021: Installation failed ...

cocopop1965, 2017-08-21:
the error message said to report here ...

Installation failed !!

Last login: Mon Aug 21 15:58:44 on ttys000
Mac-Book-Pro:~ George$ python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"

==> Installing Python Package Manager ...
pip 9.0.1 from /Library/Python/2.7/site-packages (python 2.7)

[SUCCESS]

==> Installing PlatformIO and dependencies ...

/usr/bin/python: No module named --no-cache-dir

[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio-core/issues >
Mac-Book-Pro:~ George$

ivankravets, 2017-08-23:
Please provide an output from system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

cocopop1965, 2017-08-23:
Hi Ivankravits, what command do I enter in terminal to get this information ?

ivankravets, 2017-08-23:
Hi,

See a list with commands above.

cocopop1965, 2017-08-23:
Ok thanks I figured they were the commands , I should have tried them before I asked .i was lazy as I  just woke up :-)   Cheers

cocopop1965, 2017-08-23:
Hmm, ok I tried those commands and that have me no info ?

cocopop1965, 2017-08-23:
Python 2.7.10
pip 9.0.1
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
altgraph (0.10.2)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
bottle (0.12.13)
certifi (2017.7.27.1)
chardet (3.0.4)
idna (2.6)
lockfile (0.12.2)
macholib (1.5.1)
matplotlib (1.3.1)
modulegraph (0.10.4)
numpy (1.8.0rc1)
pip (9.0.1)
py2app (0.7.3)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
python-dateutil (1.5)
pytz (2013.7)
requests (2.18.4)
scipy (0.13.0b1)
setuptools (18.5)
six (1.4.1)
urllib3 (1.22)
wheel (0.29.0)
xattr (0.6.4)
zope.interface (4.1.1)

ivankravets, 2017-08-24:
Could you try in system terminal `sudo pip install platformio`?

cocopop1965, 2017-08-26:
That worked Thanks !!  I appreciate your help

ivankravets, 2017-08-26:
Thank you too for a report!

closed 2017-08-26 by ivankravets