platformio/platformio-core #1022: pre build extra script is not behaving as expected or documentation missing

rahuldeo2047, 2017-08-22:
I need to update a version.h file using git cmd.
Need to execute relevant script part to update version.h file before the build starts so that the git version is updated and goes into the firmware.
Please let me know if I have missed something important.

- [ ] PlatformIO Core. 

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.0a4

### Description of problem
The script in 'pre' in extra_script should execute before the code is built. Which is not happening.

#### Steps to Reproduce

1. file.py:
```
Import("env")

def pre_build(source, target, env):
    print "################pre_build"
    # do some actions

env.AddPreAction("buildprog", pre_build)
```
2. This file is in the level of platformio.ini
3. Used `extra_scripts = file.py` following http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions

### Actual Results
1. When I add 'pre' there is no printing as per the extra_script, but when there is nothing in place of pre it prints is after all build is done. Which is documented as 'post' by default.

### Expected Results
1. Putting 'pre' should result in executing the file before the build is started so that my version.h file is updated before the build.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp01]
platform = espressif8266
framework = arduino
board = d1
extra_scripts = pre:file.py
```
actual file name : pre_extra_scripts.py
 
### Additional info
1. I also tried with / without pre: but saving, restarting ATOM did not help me out.
2. Tried with 'buildfs' as well. env.AddPreAction("buildfs", pre_build) 
3. e.g. version file contents:
```
#ifndef VERSION_H
#define VERSION_H

#define _VER_ ( "v0.0.2-7-g78b38a2" )

#endif //VERSION_H
```
4. shell script which generates it 
```
    versionfile='version.h'
    gitver="$(git describe --tags --long)"
    echo '#ifndef VERSION_H' > $versionfile
    echo '#define VERSION_H' >> $versionfile
    echo '' >> $versionfile
    echo "#define _VER_ ( \"$gitver\" )" >> $versionfile
    echo '' >> $versionfile
    echo '#endif //VERSION_H' >> $versionfile
```

rahuldeo2047, 2017-08-23:
Evaluating http://docs.platformio.org/en/latest/projectconf/section_env_build.html

rahuldeo2047, 2017-08-29:
In platformio.ini file

 `build_flags = !echo "-DVER="$(git describe --tags --long)`

In aheader.h file

```

#define STR(x) #x
#define XSTR(x) STR(x) 
#define _VER_ XSTR(VER)

```
have done the trick.

closed 2017-08-29 by rahuldeo2047

corbolais, 2018-12-22:
I think this should be reopened.

The solution presented by @rahuldeo2047 is a workaround, not a fix. At least not in the sense of the misbehaviour  of `pre:` being corrected.

Having tried to implement the same idea, I have come to the same conclusion. Namely, the `pre:` semantics are still broken, if they ever worked as intended and documented at all.

It is impossible to implement such logic w/ the present `platformio` version:

    pio --version
    PlatformIO, version 3.6.3

cheers

martipops, 2025-04-15:
I would agree this should probably be reopened or be fixed in the documentation. 

It's definitely a headache to workaround when it would be preferable if `pre:script.py` would function BEFORE the build process. Most use cases will be for generating helper files or header files for use in the project's src. (For example `xxd` to compile files into byte arrays).