platformio/platformio-core #1029: How to add Jlink debug tool

wofei1314, 2017-09-02:
I have sad this page
http://docs.platformio.org/en/latest/plus/debugging.html#custom

But I didn't add Jlink successful.

My platformio ide version is 0.7.2

I want to use PIO Debug my stm32 project with Jlink debug tool ...

I try to modify platformio.ini like this:
[env:jlink_debug_and_upload]
platform = ststm32
framework = stm32cube
board = nucleo_f411re
debug_tool = custom
debug_server =
  JLinkGDBServerCL
  -singlerun
  -if
  SWD
  -select
  USB
  -port
  2331
  -device
  STM32F411RE

And when I update firmware to mcu,

`[env:jlink_debug_and_upload]
platform = ststm32
framework = stm32cube
board = nucleo_f411re
debug_tool = custom
debug_server =
  JLinkGDBServerCL
  -singlerun
  -if
  SWD
  -select
  USB
  -port
  2331
  -device
  STM32F411RE`

------------------------------------------------------------------

### Configuration

**Operating system**:WIN10 64bit

**PlatformIO Version** (`platformio --version`):0.7.2

ivankravets, 2017-09-03:
Could you try this workaround?

http://docs.platformio.org/en/latest/plus/debugging.html#j-link-as-debugger-and-uploader

If it works, we will think how to integrate J-Link to our development platforms where user will use something like that:

```
[env:myenv]
...
debug_tool = jlink
upload_protocol = jlink
```

medovarsky, 2017-10-02:
Hi Ivan,

Confirming that extra_script.py works as expected in upload and debug modes. Tested with "JLinkExe -device STM32L073VZ" using STM32L073VZ-EVAL board.

Target upload:
`
Uploading .pioenvs/eval_l073z/firmware.bin
2017-10-02T17:21:03 INFO src/common.c: Loading device parameters....
2017-10-02T17:21:03 INFO src/common.c: Device connected is: L0x Category 5 device, id 0x20086447
2017-10-02T17:21:03 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x30000 bytes (192 KiB) in pages of 128 bytes
2017-10-02T17:21:03 INFO src/common.c: Attempting to write 12280 (0x2ff8) bytes to stm32 address: 134217728 (0x8000000)
st-flash 1.3.1
2017-10-02T17:21:04 INFO src/common.c: Finished erasing 96 pages of 128 (0x80) bytes
2017-10-02T17:21:04 INFO src/common.c: Starting Half page flash write for STM32L core id
2017-10-02T17:21:04 INFO src/flash_loader.c: Successfully loaded flash loader in sram
Flash page at addr: 0x08002f80 erased
2017-10-02T17:21:05 INFO src/common.c: Starting verification of write complete
2017-10-02T17:21:05 INFO src/common.c: Flash written and verified! jolly good!
191/191 halfpages written
============================================================================ [SUCCESS] Took 6.40 seconds ============================================================================
--- Miniterm on /dev/ttyACM0  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---`
`

ju-mpe-r, 2018-02-10:
with
```
upload_protocol = jlink
debug_tool = custom
debug_server =
    JLinkGDBServerExe
    -singlerun
    -if
    SWD
    -select
    USB
    -port
    2331
    -device
    STM32F103C8
    -telnetport
    2333
```
I get
```
AVAILABLE: blackmagic, jlink, jlink, serial, stlink
CURRENT: upload_protocol = jlink
Uploading .pioenvs/genericSTM32F103C8/firmware.bin
2018-02-11T00:48:23 WARN usb.c: Couldn't find any ST-Link/V2 devices
st-flash 1.4.0
*** [upload] Error 255
```

ju-mpe-r, 2018-02-10:
but when I push Debug button uploading finished fine

ju-mpe-r, 2018-02-11:
my extra_script.py
```
Import("env")

env.AddPreAction(
    "upload",
     env.VerboseAction(" ".join([
         "echo",
         "'h\\nloadfile $BUILD_DIR/firmware.hex\\nr\\nq\\n'",
         ">$BUILD_DIR/aux.jlink"
     ]), "Creating auxiliary files"))

env.Replace(
    UPLOADHEXCMD=
    'JLinkExe -device STM32F103C8 -speed 4000 -if swd -autoconnect 1 -CommanderScript $BUILD_DIR/aux.jlink'
)
```

ju-mpe-r, 2018-02-11:
So in my case I solved this issue with customizing extra_script.py
```
Import("env")

env.AddPreAction(
    "upload",
     env.VerboseAction(" ".join([
         "echo -e",
         "'h\\nloadbin $BUILD_DIR/firmware.bin 0x8000000\\nr\\nq\\n'",
         ">$BUILD_DIR/aux.jlink",
     ]), "Creating auxiliary files"))

env.Replace(
    UPLOADCMD = 'JLinkExe -device STM32F103C8 -speed 4000 -if swd -autoconnect 1 -CommanderScript $BUILD_DIR/aux.jlink'
)
```

ivankravets, 2018-02-19:
@ju-mpe-r could you re-test with the latest upstream version?  http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

Please remove all extra scripts and lines.

Just use 

```
[env:xxx]
upload_protocol = jlink
debug_tool = jlink
```

All should work out of the box.

ivankravets, 2018-02-19:
@wofei1314 please also re-test with http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

closed 2018-02-19 by ivankravets

ju-mpe-r, 2018-02-27:
@ivankravets 
I got this
`[Tue Feb 27 13:13:25 2018] Processing genericSTM32F103C8 (platform: https://github.com/platformio/platform-ststm32.git; board: genericSTM32F103C8; framework: stm32cube)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > STM32F103C8 (20k RAM. 64k Flash)
SYSTEM: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DebugInvalidOptions: Unknown debug tool `jlink`. Please use one of `blackmagic, stlink-v2, stlink-v2-1` or `custom`:
File "/home/jj/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 176:
env.SConscript("$BUILD_SCRIPT")
File "/home/jj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/jj/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/home/jj/.platformio/platforms/ststm32@src-15e0e87f41a00aa2f9602f71367ac29c/builder/main.py", line 193:
target_elf = env.BuildProgram()
File "/home/jj/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/jj/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 45:
env.PrintConfiguration()
File "/home/jj/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/jj/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/pioplatform.py", line 134:
env.subst("$DEBUG_TOOL"))
File "/home/jj/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 731:
(tool_name, ", ".join(sorted(debug_tools.keys()))))
============================================================= [ERROR] Took 0.23 seconds =============================================================
The terminal process terminated with exit code: 1
`

ju-mpe-r, 2018-02-27:
@ivankravets 
with ```transport select swd``` in ```.platformio/packages/tool-openocd/scripts/interface/jlink.cfg``` works fine :)

ivankravets, 2018-03-02:
@ju-mpe-r I've just fixed that and you don't need to modify openOCD scripts. Could you try the upstream version?

- http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

jourjine, 2018-03-12:
[env:upstream]
platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8
framework = arduino
upload_protocol = jlink
debug_tool = jlink
debug_port =localhost:2331

Cloning into 'C:\Users\Jourjine\.platformio\platforms\_tmp_installing-erzqj2-package'...
fatal: unable to access 'https://github.com/platformio/platform-ststm32.git/': error setting certificate verify locations:
  **CAfile: C:/Program Files/Git/mingw64/libexec/ssl/certs/ca-bundle.crt**
  CApath: none
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/platformio/platform-ststm32.git', 'C:\\Users\\Jourjine\\.platformio\\platforms\\_tmp_installing-erzqj2-package']

mistake path -  C:/Program Files/Git/mingw64/libexec/ssl/certs/ca-bundle.crt
correct path - C:/Program Files/Git/mingw64/ssl/certs/ca-bundle.crt

ivankravets, 2018-03-12:
@jourjine you have broken Git tool in a system. Try to reinstall it or use `platform = ststm32`

ju-mpe-r, 2018-03-21:
@ivankravets yes, it's working fine right now.

ju-mpe-r, 2018-03-21:
@ivankravets, I was wrong. Successful result was with my `extra_script.py` in `platformio.ini`, but without it, I've got this:
```GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-14:56)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
swd
Warn : Transport "swd" was already selected
swd
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Error: Failed to claim interface: LIBUSB_ERROR_BUSY.
Error: Failed to open device: unspecified error.
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured.
*** [upload] Error 255```

ivankravets, 2018-03-22:
@ju-mpe-r could you close VSCode, disconnect J-Link and start again?
Of course, without extra script.

ju-mpe-r, 2018-03-22:
@ivankravets working! Thank you!