platformio/platformio-core #1032: Add support M5Stack that's used ESP32

sczh01, 2017-09-03:
Formal package: http://www.m5stack.com/download/package_m5stack_index.json

ivankravets, 2017-09-05:
Resolved in https://github.com/platformio/platform-espressif32/releases/tag/v0.10.0

Please `pio update`.

closed 2017-09-05 by ivankravets

sczh01, 2017-09-05:
HI Ivankravets,

Thanks for your support!
Just update PIO, below is some error for your reference:
-------------------------------------------------------------------------------------------
Updating espressif32                     @ 0.9.1          [0.10.0]
Uninstalling espressif32 @ 0.9.1:       Error: Traceback (most recent call last):
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\tony\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\tony\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\tony\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\tony\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\tony\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\tony\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\commands\update.py", line 46, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "c:\users\tony\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 376, in platform_update
    platform, only_packages=only_packages, only_check=only_check)
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 120, in update
    BasePkgManager.update(self, pkg_dir, requirements, only_check)
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 735, in update
    self.uninstall(pkg_dir, trigger_event=False)
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 92, in uninstall
    BasePkgManager.uninstall(self, pkg_dir, requirements)
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 672, in uninstall
    util.rmtree_(pkg_dir)
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\util.py", line 636, in rmtree_
    return rmtree(path, onerror=_onerror)
  File "C:\Users\tony\.platformio\python27\Lib\shutil.py", line 247, in rmtree
    rmtree(fullname, ignore_errors, onerror)
  File "C:\Users\tony\.platformio\python27\Lib\shutil.py", line 256, in rmtree
    onerror(os.rmdir, path, sys.exc_info())
  File "c:\users\tony\.platformio\penv\lib\site-packages\platformio\util.py", line 634, in _onerror
    os.remove(name)
WindowsError: [Error 5] : 'C:\\Users\\tony\\.platformio\\platforms\\espressif32\\boards'

ivankravets, 2017-09-05:
Please remove manually this folder `C:\Users\tony\.platformio\platforms\espressif32`

sczh01, 2017-09-05:
I have remove the this folder manually and can "PIO update" can work.
And PIO home still can't been connectted. 
--------------------------------------------------------------
“PIO Home”: Could not start PIO Home server: Error: timeout。

ivankravets, 2017-09-05:
Which IDE do you use? Try to close all IDEs, and kill all `platformio.exe` and `pioplus.exe` processes. Start IDE.

sczh01, 2017-09-05:
Thanks, I kill pioplus.exe manually, then PIO home can connect.
I use the VSC-1.15.1 in win7/32 bit/professinal version.

sczh01, 2017-09-05:
Sorry, I still can't build the project when I create a new project by this Boards.
I place the some file into lib dir.

----------------------------------------------------------------------------------------------

Compiling .pioenvs\m5stack-core-esp32\src\main.o
Generating partitions .pioenvs\m5stack-core-esp32\partitions_table.bin
<command-line>:0:16: warning: ISO C99 requires whitespace after the macro name
src\main.c:1:21: fatal error: M5Stack.h: No such file or directory
[170906-061537-m5stack-core-esp32.zip](https://github.com/platformio/platformio-core/files/1279205/170906-061537-m5stack-core-esp32.zip)

compilation terminated.
*** [.pioenvs\m5stack-core-esp32\src\main.o] Error 1

ivankravets, 2017-09-05:
Please file an issue here https://github.com/platformio/platform-espressif32/issues