platformio/platformio-core #1033: PIO Home: Feature requests

ivankravets, 2017-09-03:
# Project Wizard

- [ ] "Update" check button would be helpful, which gives the actual project path like before ( @sstaub )
- [ ] PIO Home -> Clone function for projects would be nice. ( @sstaub )
- [ ] Ability to "star" any folder and see it in file explorer "Places" section.  ( @sstaub )

neilmendoza, 2017-11-07:
The new file browser for "non-default location" project creation is super clunky. Any reason for not using the system file browser?

ivankravets, 2017-11-07:
@neilmendoza you can share PIO Home with other developers within a local network. Also, remote access to PIO Home is coming soon. PIO Home could be started on a server and used from other machines.

Nevertheless, we will improve current FileExplorer with new features, such as an ability to mark a folder as a favourite, create a new or rename existing.

ivankravets, 2017-12-03:
File Explorer has been refactored in 3.5.0rc5. Please `pio upgrade`.

ivankravets, 2017-12-04:
Implemented in PIO Home 0.5.0. Please update

closed 2017-12-04 by ivankravets