platformio/platformio-core #1035: Mongoose OS support

OPatrick, 2017-09-04:
Please add support for Mongoose OS as it is very quick to develop with Espressif 32 and others.

Thanks

di-ai-wai, 2017-11-01:
Are there any plans for supporting mangoose?

mamuesp, 2018-04-19:
No answers? Hm ... I would like it so.

ivankravets, 2018-04-19:
Which environment do you use currently working with Mongoose OS?

/cc @cpq

i5okie, 2018-12-12:
not anymore it doesn't.
they have some experimental VSCode plugin.
And their awesome webui is now a meh ui since version 2.7.0. no more built-in ide.

NeliusNDL, 2020-10-24:
Mongoos OS community dead ?

cpq, 2020-10-27:
@NeliusNDL it is not

NeliusNDL, 2020-10-27:
> @NeliusNDL it is not

Excellent. Please add my vote

NeliusNDL, 2020-10-27:
> > @NeliusNDL it is not
> 
> Excellent. Please add my vote

@cpq maybe you can give us an example program for platform IO using ESP IDF framework. It is difficult to get all the libraries together and compiling in PIO

cpq, 2020-10-29:
@NeliusNDL , unfortunately I cannot provide such example. I am affiliated with neither PlatformIO, nor with ESP IDF. I am affiliated with Mongoose OS, and can speak for it.

NeliusNDL, 2020-10-29:
Ok thanks