platformio/platformio-core #1039: Home: Could not load platform data

paulbearne, 2017-09-05:
PIO Core Call Error: got stderr: "Error: Unknown development platform 'espressif32@0.8.0'\r\n"

ivankravets, 2017-09-05:
Could you close and reopen PIO Home? Do you still see this error?

paulbearne, 2017-09-05:
done but getting multiple errors it appears that when i installed the atom ide and its platform io i can no longer run the visual studio code version keeps coming up with the above error as well as a host of python errors when ever i try to compile also if i close and reopen vsc i get Unable to open 'PIO Home': Could not start PIO Home server: Error: timeout. i'm using vsc 1.15.1 insider but up until i installed the atom it was all working fine

ivankravets, 2017-09-05:
1. Please close all IDEs
2. Open OS Task Manager and kill all python/pioplus processes.

It should help.

paulbearne, 2017-09-05:
already tried that and did a full reboot just to be sure 
just to verify the problem only occurred after i installed the atom IDE.
if run eclipse and compile and flash using the esp-idf works fine arduino ide works fine 
atom ide slow and sluggish but works and compiles but now my my VSC comes up with errors 
i have uninstalled atom but it leaves crap everywhere doing a reg clean now 

Python errors
Exception in thread Thread-2:
Traceback (most recent call last):
  File "c:\msys32\mingw32\lib\python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\msys32\mingw32\lib\python2.7\site-packages/platformio/util.py", line 92, in run
    self.outcallback(line)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/platformio/managers/platform.py", line 381,
in on_run_out
    self._echo_line(line, level=1)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/platformio/managers/platform.py", line 396,
in _echo_line
    click.secho(line, fg=fg, err=level > 1)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/click/termui.py", line 417, in secho
    return echo(style(text, **styles), file=file, nl=nl, err=err, color=color)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/click/utils.py", line 314, in echo
    file.write(message)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/click/_compat.py", line 560, in _safe_write
    return _write(s)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/colorama/ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/colorama/ansitowin32.py", line 141, in write

    self.write_and_convert(text)
  File "c:\msys32\mingw32\lib\python2.7\site-packages/colorama/ansitowin32.py", line 169, in write
_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "c:\msys32\mingw32\lib\python2.7\site-packages/colorama/ansitowin32.py", line 174, in write
_plain_text
    self.wrapped.write(text[start:end])
  File "c:\msys32\mingw32\lib\python2.7\site-packages/click/_compat.py", line 97, in write
    return self.buffer.write(str(x))
IOError: [Errno 0] Error

ivankravets, 2017-09-05:
What is your
```
 pio --version
```?

Please open PlatformIO IDE Terminal in VSCode (terminal button on the bottom left corner) and type `pio upgrade`. Does it work now?

paulbearne, 2017-09-05:
3.5.0a9 already tried upgrade tells me everything is latest version also run pio update to make sure the framework is all up to date which it is another  thing might help if i reinitialize the project dir using the coap server example to test it then runs through compiling and linking as it should but then continues by recompiling and linking again and again stuck in some loop until i kill vsc then after restarting all the python errors are back

ivankravets, 2017-09-05:
The last issue was fixed in 3.5.0a8 ( https://github.com/platformio/platformio-core/commit/a127251107afe133dc50ff7e49f4337c7688413d )

Which errors do you see now?

paulbearne, 2017-09-05:
The same no change i'm going to uninstall it all when i get time and give it another go it seems that something got installed when i put the atom ide on that has messed up my exisiting install remember the vcs version was always there been there for months just updating when it said one was ready until i installed the atom ide version this morning

ivankravets, 2017-09-05:
Could you try a clean uninstalling? 
- http://docs.platformio.org/en/latest/ide/atom.html#uninstall-atom-with-platformio-ide

paulbearne, 2017-09-06:
Trying that now i take when they say they just mean anything with platformio or atom 
Remove registry records using
HKEY_CLASSES_ROOT\Directory\Background\shell
HKEY_CLASSES_ROOT\Directory\shell
HKEY_CLASSES_ROOT*\shell

ivankravets, 2017-09-06:
Sorry, I can't understand. Which IDE do you use: Atom or VSCode?

paulbearne, 2017-09-06:
VSCode it was installing atom that screwed everything up finally managed to get it to work after wading through the registry and my hard drive to find everything that it had installed and then did a fresh install which is now working think someone need to take a serious look at the uninstaller as it didn't uninstall anything but the executable

ivankravets, 2017-09-06:
Do you mean that all works well now?

paulbearne, 2017-09-06:
Yes Thankyou

closed 2017-09-06 by ivankravets

jamesbiederbeck, 2018-01-29:
paul, any tips on where to start with the registry?

ivankravets, 2018-01-29:
@jamesbiederbeck which type of problem do you have?