platformio/platformio-core #1040: Hard hang on install/upgrade of Platformio-Core

madscientist42, 2017-09-05:
OS: Windows 7, 64-bit
Version: Unknown.  (Latest?? Couldn't tell you because it won't finish because of the problem encountered...)

Upgrade/Install ends with a fatal error dialog, _*overlaid*_ by your "Installing" dialog with no way to acknowlege or fix the issue.  Fatal error dialog: "'platformio' is not recognized as an iternal or external command, operable program, or batch file.  Error: spawn platformio ENOENT"

Problem is twofold.

1) You take total control of Atom, rendering non-operable when you fail like this.  I'm going to have to rip the whole thing kicking and screaming out and fall back to the vendor platform build framework (Thanks, Guys!!)

2) This strikes me as sloppy- it appears that in the middle of the download, you got rid of the core to make way for the new copy and something pulled the ripcord on needing the new core install, which only is handled by _*your*_ application code that cripples Atom here.  It doesn't feel like you even tested this before rolling it out. 

It doesn't do any good to support a whole raftload of platforms when it breaks down like this and you take the whole use of Atom away from people (Heh...I use Atom for a LOT more than your project/product.  Scala, Haskell, few others...)

ivankravets, 2017-09-05:
Could you try to remove `%HOMEDIR%/.platformio/penv` folder and restart?

nmaas87, 2017-09-05:
I had the same problem and did not have virtualenv installed.
Trying installing it via 
`python -m pip install virtualenv`
and then restart Atom again :)

ivankravets, 2017-09-05:
Fixed in https://github.com/platformio/platformio-atom-ide/issues/906

closed 2017-09-05 by ivankravets