platformio/platformio-core #1043: 2.0.0-rc.1: loop after uncheck "Use build-in PlatformIO Core"

rwalli, 2017-09-06:
Also tried a fresh install on macOS 10.12.6 / Atom 1.19.5:
cd ~
rm -fr .atom
rm -fr .platformio
atom

install platformio-ide in atom
-> Restart
uncheck "Use build-in PlatformIO Core" in platformio settings

This message loops after each restart of Atom:
-----------------------------------------------
PlatformIO IDE has been successfully installed!
Please restart Atom to apply the changes
-----------------------------------------------

ivankravets, 2017-09-06:
Please move issue to https://github.com/platformio/platformio-atom-ide/issues

closed 2017-09-06 by ivankravets

rwalli, 2017-09-06:
uhhhps, sorry for the wrong place, don't know ho to move this issue.
should i open a new one in the right place?

ivankravets, 2017-09-06:
Yes, please open new :)