platformio/platformio-core #1047: Teesny 3.6: Doesn't work without Arduino Framework

jonathanrjpereira, 2017-09-07:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`):

### Description of problem
I want to program the Teensy 3.6 using bare metal embedded C without using the Arduino Libraries.
I have included the header files 'kinetis.h' & 'core_pins.h' for this purpose.

After `pio run`, I get this error:

```
 .pioenvs\teensy36\libFrameworkArduino.a(main.o): In function `main':
C:\Users\Jonathan\.platformio\packages\framework-arduinoteensy\cores\teensy3/main.cpp:21: undefined reference to `setup'
C:\Users\Jonathan\.platformio\packages\framework-arduinoteensy\cores\teensy3/main.cpp:23: undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\teensy36\firmware.elf] Error 1
```
The problem is solved if I use the '`void loop`' & '`void setup`'. But I do not want to use Arduino syntax.

#### Steps to Reproduce

1. Use Teensy 3.6 as board
2. The src code shouldn't contain the setup & loop.

### Actual Results
Compile & Upload successful

### Expected Results
```
 .pioenvs\teensy36\libFrameworkArduino.a(main.o): In function `main':
C:\Users\Jonathan\.platformio\packages\framework-arduinoteensy\cores\teensy3/main.cpp:21: undefined reference to `setup'
C:\Users\Jonathan\.platformio\packages\framework-arduinoteensy\cores\teensy3/main.cpp:23: undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\teensy36\firmware.elf] Error 1
```

ivankravets, 2017-09-07:
Did you remove `framework` option from `platformio.ini`?

jonathanrjpereira, 2017-09-07:
no

ivankravets, 2017-09-07:
if you don't want to use Arduino framework for Teensy, you have must remove `framework` option

closed 2017-09-07 by ivankravets