platformio/platformio-core #1048: Home: Could not list directory/home/efischer

feugen, 2017-09-08:
Could not list directory by `/home/xyz`

Hello guys, fresh install of atom and platformio. Home 0.2.5·Core 3.5.0a9

Problem 1: I cant select a custom directory and get an error when I try to open home directory : "Could not list directory by `/home/xyz`". Its not a permission problem, I checked it and the permission is set correctly.

Problem 2: If I try to start a new project, the default project directory is not localized, so it creates a "documents" directory instead of using german "Dokumente" directory, but it is at least able to create the folder.

ivankravets, 2017-09-09:
Is it a real error with `/home/xyz`?

feugen, 2017-09-09:
No, it says `/home/efischer` and its my home folder

ivankravets, 2017-09-09:
What is your locale?

feugen, 2017-09-09:
```
LANG=de_DE.utf8
LANGUAGE=
LC_CTYPE="de_DE.utf8"
LC_NUMERIC="de_DE.utf8"
LC_TIME="de_DE.utf8"
LC_COLLATE="de_DE.utf8"
LC_MONETARY="de_DE.utf8"
LC_MESSAGES="de_DE.utf8"
LC_PAPER="de_DE.utf8"
LC_NAME="de_DE.utf8"
LC_ADDRESS="de_DE.utf8"
LC_TELEPHONE="de_DE.utf8"
LC_MEASUREMENT="de_DE.utf8"
LC_IDENTIFICATION="de_DE.utf8"
LC_ALL=
```

ivankravets, 2017-09-09:
Which IDE do you use?

ivankravets, 2017-09-09:
Please try to open system terminal (not terminal from IDE) and type:
```
~/.platformio/penv/bin/platformio home
```

It should open PIO Home in browser. Does it work?

feugen, 2017-09-09:
I am using Atom 1.19.7
I have tried `~/.platformio/penv/bin/platformio home` and it opens in browser tab (address http://127.0.0.1:8008/). I am now able to browse my home directory and I have successfully created a new project. It looks like Atom is able to open this project.

ivankravets, 2017-09-09:
Thanks a lot! Fixed in https://github.com/platformio/platformio-node-helpers/releases/tag/v0.2.4

For Atom: just uninstall `platformio-ide` package and install again. It will use the new updated helpers.
For VSCode: update our `platformio-ide` extension to 0.7.3.

closed 2017-09-09 by ivankravets

feugen, 2017-09-09:
Thanks, it works now. :-)

ivankravets, 2017-09-09:
Would be thankful for a  star ⭐️ https://github.com/platformio/platformio-core/stargazers 😊 

Happy coding with PlatformIO!

feugen, 2017-09-09:
No problem, I am following now :-)