platformio/platformio-core #1050: espressif8266_stage and upload files "Could not find LD script"

nardev, 2017-09-08:
I had some issues with building one project because it required latest Arduino core.

I found here that i should add "espressif8266_stage" and get it from git. Build was ok after that but uploading files failed.

```bash
$ platformio run -v --target buildfs && platformio run --target uploadfs
[Fri Sep  8 23:26:16 2017] Processing nodemcu (platform: espressif8266_stage; build_flags: -Wl,-Tesp8266.flash.4m.ld; upload_port: /dev/ttyUSB3; board: nodemcu; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Could not find LD script
======================================================================================== [ERROR] Took 0.34 seconds ========================================================================================
```

ivankravets, 2017-09-09:
Fixed in https://github.com/platformio/platform-espressif8266/issues/37

Be sure that you use the latest version of dev/platform. Run `pio update`.

jeremypoulter, 2017-09-15:
I think I am seeing the same issue running the latest staging from Git, see https://travis-ci.org/jeremypoulter/ESP8266_WiFi_v2.x/jobs/275869816 (ignore the ESPAsyncTCP errors).

There is an explicit call to `platformio update` so I am fairly sure a cached version is not in use.

ivankravets, 2017-09-15:
@jeremypoulter 

1. [Clear Travis.CI cache](https://docs.travis-ci.com/user/caching/#Clearing-Caches)
2. Replace `https://github.com/me-no-dev/ESPAsyncWebServer.git` with `ESPAsyncWebServer` [here](https://github.com/jeremypoulter/ESP8266_WiFi_v2.x/blob/install_improvements/platformio.ini#L84)

Restart build

jeremypoulter, 2017-09-15:
Still the same for this issue (and thanks for pointing out the obvious fix for the ESPAsyncWebServer issue)

ivankravets, 2017-09-15:
@jeremypoulter the error is very strange. Could you fix `lib_deps`?

jeremypoulter, 2017-09-15:
Yeah, done already.

ivankravets, 2017-09-16:
Thanks! Fixed in https://github.com/platformio/platform-espressif8266/commit/43976990564556cca9c9e197f8bf16674dad1fba

Just restart a build.

closed 2017-09-16 by ivankravets

jeremypoulter, 2017-09-16:
Yeah, that fixed it, thanks.