platformio/platformio-core #1053: Home: Internal Store Exception

apenaz, 2017-09-09:
TypeError: Cannot read property 'filter' of null
    at t.value (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:86699)
    at http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:86385
    at Array.map (native)
    at t.value (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:86240)
    at t.value (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:84312)
    at d._renderValidatedComponentWithoutOwnerOrContext (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:528221)
    at d._renderValidatedComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:528348)
    at d._updateRenderedComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:527652)
    at d._performComponentUpdate (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:527453)
    at d.updateComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:526731)

ivankravets, 2017-09-09:
How to reproduce this issue?

ivankravets, 2017-09-15:
Duplicate of https://github.com/platformio/platformio-core/issues/1059

closed 2017-09-15 by ivankravets