platformio/platformio-core #1054: Home: Could not load boards list

S-Harshow, 2017-09-09:
PIO Core Call Error: got stderr: "sh: mc: line 1: syntax error: unexpected end of file\nsh: error importing function definition for `mc'\n"

ivankravets, 2017-09-09:
Could you open PlatformIO IDE Terminal and type:
```
pio boards --json-output
```

Do you see some text?

S-Harshow, 2017-09-09:
В письме от суббота, 9 сентября 2017 г. 15:57:25 +05 пользователь Ivan Kravets 
написал:
> Could you open PlatformIO IDE Terminal and type:
> ```
> pio boards --json-output
> ```
> 
> Do you see some text?
Yes, there is a lot of text, like this:
[{"vendor": "4DSystems", "rom": 520192, "frameworks": ["arduino"], "url": 
"http://www.4dsystems.com.au/product/Picadillo_35T/", "fcpu": 80000000, "ram": 
131072, "connectivity": null, "platform": "microchippic32", "debug": null, 
"mcu": "32MX795F512L", "id": "picadillo_35t", "name": "4DSystems PICadillo 
35T"}, {"vendor": "96Boards", "rom": 524288, "frameworks": ["mbed", 
"stm32cube"], "url": "https://developer.mbed.org/platforms/ST-B96B-F446VE/", 
"fcpu": 168000000, "ram": 131072, "connectivity": ["can"], "platform": 
"ststm32", "debug": {"tools": {"stlink-v2-1": {"onboard": true}}}, "mcu": 
"STM32F446VET6", "id": "b96b_f446ve", "name": "96Boards B96B-F446VE"}, 
{"vendor": "NXP", "rom": 32768, "frameworks": ["mbed"], "url": "https://
developer.mbed.org/handbook/mbed-NXP-LPC11U24", "fcpu": 48000000, "ram": 8192, 
"connectivity": null, "platform": "nxplpc", "debug": {"tools": {"cmsis-dap": 
{"onboard": true}}}, "mcu": "LPC11U24", "id": "lpc11u24_301", "name": "ARM 
mbed LPC11U24 (+CAN)"}, {"vendor": "Adafrui...

If in console I type "atom" there are no errors, so I change in *.desktop file 
command to "atom %F" instead of "/usr/...../atom %F".

ivankravets, 2017-09-09:
Could you try PIO IDE for VSCode?

- http://platformio.org/platformio-ide

ivankravets, 2017-09-13:
Please run in PIO IDE Terminal and provide an output:
```
python -c "import os; print os.environ"
```

ivankravets, 2017-09-13:
@domingosliraneto , @scottdarch please provide the same output of https://github.com/platformio/platformio-core/issues/1054#issuecomment-329136082

domingosliraneto, 2017-09-13:
Hi, it's okay. Thank you for your help! @ivankravets

ivankravets, 2017-09-13:
@domingosliraneto does it work for you now?

domingosliraneto, 2017-09-14:
Yes, thanks!

ivankravets, 2017-09-14:
@domingosliraneto how did you resolve that?

domingosliraneto, 2017-09-14:
The place where I work uses proxy server, for some reason it blocked the installation of the PlatiformIO dependencies, I only did the reinstallation when I got home and everything worked.

ivankravets, 2017-09-15:
@thosoo Please run in PIO IDE Terminal and provide an output:
```
python -c "import os; print os.environ"
```

S-Harshow, 2017-09-15:
В письме от пятница, 15 сентября 2017 г. 16:07:15 +05 пользователь Ivan 
Kravets написал:
> @thosoo Please run in PIO IDE Terminal and provide an output:
> ```
> python -c "import os; print os.environ"
> ```
sergey@Asgard:~/Projects/github> python -c "import os; print os.environ"
{'LESS': '-M -I -R', 'PLATFORMIO_CALLER': 'atom', 'PLATFORMIO_IDE': '2.0.0-rc.
2', 'XDG_SESSION_TYPE': 'x11', 'XKEYSYMDB': '/usr/X11R6/lib/X11/X
KeysymDB', 'KDE_FULL_SESSION': 'true', 'XSESSION_IS_UP': 'yes', 
'XDG_DATA_DIRS': '/usr/share', 'GS_LIB': '/home/sergey/.fonts', 'HISTSIZE': 
'10
00', 'LESS_ADVANCED_PREPROCESSOR': 'no', 'MANPATH': '/usr/local/man:/usr/
share/man', 'XMODIFIERS': '@im=local', 'JAVA_HOME': '/usr/lib64/jvm/jr
e', 'PROFILEREAD': 'true', 'QT_AUTO_SCREEN_SCALE_FACTOR': '0', 
'XDG_RUNTIME_DIR': '/run/user/1000', 'QT_IM_SWITCHER': 'imsw-multi', 
'XDG_SESSIO
N_ID': '2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:abstract=/tmp/dbus-
lcQxFrhK8N,guid=749d860388cd074b56dd6ff259bb8ce8', 'DESKTOP_SESSION': '/usr/sh
are/xsessions/plasma5', 'CONFIG_SITE': '/usr/share/site/x86_64-unknown-linux-
gnu', 'GTK_MODULES': 'canberra-gtk-module', 'HOSTNAME': 'Asgard.su
se', 'MAIL': '/var/spool/mail/sergey', 'MACHTYPE': 'x86_64-suse-linux', 
'SSH_ASKPASS': '/usr/lib/ssh/ksshaskpass', 'JAVA_ROOT': '/usr/lib64/jvm
/jre', 'MINICOM': '-c on', 'CSHEDIT': 'emacs', 'LESSOPEN': 'lessopen.sh %s', 
'CVS_RSH': 'ssh', 'USER': 'sergey', 'INPUTRC': '/home/sergey/.inpu
trc', 'XDG_VTNR': '7', 'XAUTHORITY': '/tmp/xauth-1000-_0', 'LANGUAGE': 'ru', 
'SESSION_MANAGER': 'local/Asgard.suse:@/tmp/.ICE-unix/2707,unix/As
gard.suse:/tmp/.ICE-unix/2707', 'SHLVL': '3', 'ALSA_CONFIG_PATH': '/etc/alsa-
pulse.conf', 'CPU': 'x86_64', 'GPG_AGENT_INFO': '/tmp/gpg-dGOL9u/S
.gpg-agent:2646:1', 'XDG_SESSION_DESKTOP': 'KDE', 'JAVA_BINDIR': '/usr/lib64/
jvm/jre/bin', 'SHELL': '/bin/bash', 'GTK2_RC_FILES': '/etc/gtk-2.0
/gtkrc:/home/sergey/.gtkrc-2.0:/home/sergey/.config/gtkrc-2.0', 
'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'LS_OPTIONS': '-N --
c
olor=tty -T 0', 'XNLSPATH': '/usr/share/X11/nls', '_': '/home/
sergey/.platformio/penv/bin/python', 'GTK_IM_MODULE': 'cedilla', 'LS_COLORS': 
'no
=00:fi=00:di=01;34:ln=00;36:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=41;33;01:ex=00;32:*.cmd=00;32:*.exe=01;32:*.com=01;32:*.bat=0

1;32:*.btm=01;32:*.dll=01;32:*.tar=00;31:*.tbz=00;31:*.tgz=00;31:*.rpm=00;31:*.deb=00;31:*.arj=00;31:*.taz=00;31:*.lzh=00;31:*.lzma=00;31:*.zip
=00;31:*.zoo=00;31:*.z=00;31:*.Z=00;31:*.gz=00;31:*.bz2=00;31:*.tb2=00;31:*.tz2=00;31:*.tbz2=00;31:*.xz=00;31:*.avi=01;35:*.bmp=01;35:*.fli=01;
35:*.gif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mng=01;35:*.mov=01;35:*.mpg=01;35:*.pcx=01;35:*.pbm=01;35:*.pgm=01;35:*.png=01;35:*.ppm=01;35:*.tga=0
1;35:*.tif=01;35:*.xbm=01;35:*.xpm=01;35:*.dl=01;35:*.gl=01;35:*.wmv=01;35:*.aiff=00;32:*.au=00;32:*.mid=00;32:*.mp3=00;32:*.ogg=00;32:*.voc=00
;32:*.wav=00;32:', 'XDG_CONFIG_DIRS': '/etc/xdg', 'WINDOWMANAGER': '/usr/bin/
startkde', 'GTK_RC_FILES': '/etc/gtk/gtkrc:/home/sergey/.gtkrc:/ho
me/sergey/.config/gtkrc', 'QEMU_AUDIO_DRV': 'pa', 'PAGER': 'less', 
'KDE_MULTIHEAD': 'false', 'QT_SYSTEM_DIR': '/usr/share/desktop-data', 'HOME'
: '/home/sergey', 'DISPLAY': ':0', 'TERM_PROGRAM': 'platformio-ide-terminal', 
'LANG': 'ru_RU.UTF-8', 'G_BROKEN_FILENAMES': '1', 'OSTYPE': 'linu
x', 'NNTPSERVER': 'news', 'XCURSOR_THEME': 'breeze_cursors', 
'G_FILENAME_ENCODING': '@locale,UTF-8,KOI8-R,CP1251', 'HOST': 'Asgard.suse', 
'GPG_
TTY': '/dev/pts/2', 'FROM_HEADER': '', 'LESSCLOSE': 'lessclose.sh %s %s', 
'PLATFORMIO_DISABLE_PROGRESSBAR': 'true', 'JRE_HOME': '/usr/lib64/jvm
/jre', 'MORE': '-sl', 'XDG_CURRENT_DESKTOP': 'KDE', 'HOSTTYPE': 'x86_64', 
'QT_IM_MODULE': 'xim', 'LOGNAME': 'sergey', 'XDG_SEAT': 'seat0', 'PAT
H': '/home/sergey/.platformio/penv/bin:/usr/local/bin:/usr/bin:/bin:/usr/bin/
X11:/usr/games:/usr/local/bin', 'KDE_SESSION_VERSION': '5', 'TERM'
: 'xterm-256color', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/
Session1', 'SDL_AUDIODRIVER': 'pulse', 'XDG_SESSION_CLASS': 'user', 'C
OLORTERM': '1', 'KDE_SESSION_UID': '1000', 'LESSKEY': '/etc/lesskey.bin', 
'LC_ALL': 'en_US.UTF-8', 'AUDIODRIVER': 'pulseaudio', 'PYTHONSTARTUP'
: '/etc/pythonstart', 'PWD': '/home/sergey/Projects/github', 
'DESKTOP_STARTUP_ID': 'Asgard.suse;1505475415;889384;2685_TIME11926735', 
'XAUTHLOC
ALHOSTNAME': 'Asgard.suse'}

thosoo, 2017-09-16:
@ivankravets 

> 
> [thaison@localhost ~]$ python -c "import os; print os.environ"
> {'PLATFORMIO_DISABLE_PROGRESSBAR': 'true', 'BASH_FUNC_module%%': '() {  eval `/usr/bin/modulecmd bash $*`', 'PLATFORMIO_CALLER': 'atom', 'BASH_FUNC_scl%%': '() {  local CMD=$1;', 'GJS_DEBUG_OUTPUT': 'stderr', 'LESSOPEN'
> : '||/usr/bin/lesspipe.sh %s', 'XDG_CURRENT_DESKTOP': 'GNOME', 'XDG_SESSION_TYPE': 'wayland', 'QT_IM_MODULE': 'ibus', 'LOGNAME': 'thaison', 'USER': 'thaison', 'PATH': '/home/thaison/.platformio/penv/bin:/usr/local/bin:/
> usr/local/sbin:/usr/bin:/usr/sbin:/home/thaison/.local/bin:/home/thaison/bin:/home/thaison/.local/bin:/home/thaison/bin:/home/thaison/.local/bin:/home/thaison/bin', 'XDG_VTNR': '2', 'HOME': '/home/thaison', 'DISPLAY': '
> :0', 'TERM_PROGRAM': 'platformio-ide-terminal', 'LANG': 'de_DE.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/bash', 'JOURNAL_STREAM': '9:31587', 'SESSION_MANAGER': 'local/unix:@/tmp/.ICE-unix/1744,unix/unix:/tmp/.ICE
> -unix/1744', 'XDG_DATA_DIRS': '/home/thaison/.local/share/flatpak/exports/share/:/var/lib/flatpak/exports/share/:/usr/local/share/:/usr/share/', 'WAYLAND_DISPLAY': 'wayland-0', 'HISTSIZE': '1000', 'GNOME_DESKTOP_SESSION
> _ID': 'this-is-deprecated', 'XMODIFIERS': '@im=ibus', 'GIO_LAUNCHED_DESKTOP_FILE_PID': '2435', 'PLATFORMIO_IDE': '2.0.0-rc.2', 'USERNAME': 'thaison', 'XDG_SESSION_DESKTOP': 'gnome', ' if [ "$CMD" ': ' "load" -o "$CMD" =
>  "unload" ]; then', 'XDG_RUNTIME_DIR': '/run/user/1000', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'GIO_LAUNCHED_DESKTOP_FILE': '/usr/share/applications/atom.desktop', 'GDMSESSION': 'gnome', 'MAIL': '/var/spool/mai
> l/thaison', 'LOADEDMODULES': '', 'XDG_SESSION_ID': '2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', '_': '/home/thaison/.platformio/penv/bin/python', 'MODULEPATH': '/etc/scl/modulefiles:/etc/scl/modulefi
> les:/etc/scl/modulefiles:/usr/share/Modules/modulefiles:/etc/modulefiles:/usr/share/modulefiles', 'DESKTOP_SESSION': 'gnome', 'HOSTNAME': 'localhost.localdomain', 'GDM_LANG': 'de_DE.UTF-8', 'MODULESHOME': '/usr/share/Mo
> dules', 'HISTCONTROL': 'ignoredups', 'SHLVL': '3', 'PWD': '/home/thaison', 'DESKTOP_STARTUP_ID': 'gnome-shell-1829-localhost.localdomain-/usr/share/atom/atom-1_TIME95382', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0
> :di=38;5;33:ln=38;5;51:mh=00:pi=40;38;5;11:so=38;5;13:do=38;5;5:bd=48;5;232;38;5;11:cd=48;5;232;38;5;3:or=48;5;232;38;5;9:mi=01;05;37;41:su=48;5;196;38;5;15:sg=48;5;11;38;5;16:ca=48;5;196;38;5;226:tw=48;5;10;38;5;16:ow=
> 48;5;10;38;5;21:st=48;5;21;38;5;15:ex=38;5;40:*.tar=38;5;9:*.tgz=38;5;9:*.arc=38;5;9:*.arj=38;5;9:*.taz=38;5;9:*.lha=38;5;9:*.lz4=38;5;9:*.lzh=38;5;9:*.lzma=38;5;9:*.tlz=38;5;9:*.txz=38;5;9:*.tzo=38;5;9:*.t7z=38;5;9:*.z
> ip=38;5;9:*.z=38;5;9:*.Z=38;5;9:*.dz=38;5;9:*.gz=38;5;9:*.lrz=38;5;9:*.lz=38;5;9:*.lzo=38;5;9:*.xz=38;5;9:*.zst=38;5;9:*.tzst=38;5;9:*.bz2=38;5;9:*.bz=38;5;9:*.tbz=38;5;9:*.tbz2=38;5;9:*.tz=38;5;9:*.deb=38;5;9:*.rpm=38;
> 5;9:*.jar=38;5;9:*.war=38;5;9:*.ear=38;5;9:*.sar=38;5;9:*.rar=38;5;9:*.alz=38;5;9:*.ace=38;5;9:*.zoo=38;5;9:*.cpio=38;5;9:*.7z=38;5;9:*.rz=38;5;9:*.cab=38;5;9:*.wim=38;5;9:*.swm=38;5;9:*.dwm=38;5;9:*.esd=38;5;9:*.jpg=38
> ;5;13:*.jpeg=38;5;13:*.mjpg=38;5;13:*.mjpeg=38;5;13:*.gif=38;5;13:*.bmp=38;5;13:*.pbm=38;5;13:*.pgm=38;5;13:*.ppm=38;5;13:*.tga=38;5;13:*.xbm=38;5;13:*.xpm=38;5;13:*.tif=38;5;13:*.tiff=38;5;13:*.png=38;5;13:*.svg=38;5;1
> 3:*.svgz=38;5;13:*.mng=38;5;13:*.pcx=38;5;13:*.mov=38;5;13:*.mpg=38;5;13:*.mpeg=38;5;13:*.m2v=38;5;13:*.mkv=38;5;13:*.webm=38;5;13:*.ogm=38;5;13:*.mp4=38;5;13:*.m4v=38;5;13:*.mp4v=38;5;13:*.vob=38;5;13:*.qt=38;5;13:*.nu
> v=38;5;13:*.wmv=38;5;13:*.asf=38;5;13:*.rm=38;5;13:*.rmvb=38;5;13:*.flc=38;5;13:*.avi=38;5;13:*.fli=38;5;13:*.flv=38;5;13:*.gl=38;5;13:*.dl=38;5;13:*.xcf=38;5;13:*.xwd=38;5;13:*.yuv=38;5;13:*.cgm=38;5;13:*.emf=38;5;13:*
> .ogv=38;5;13:*.ogx=38;5;13:*.aac=38;5;45:*.au=38;5;45:*.flac=38;5;45:*.m4a=38;5;45:*.mid=38;5;45:*.midi=38;5;45:*.mka=38;5;45:*.mp3=38;5;45:*.mpc=38;5;45:*.ogg=38;5;45:*.ra=38;5;45:*.wav=38;5;45:*.oga=38;5;45:*.opus=38;
> 5;45:*.spx=38;5;45:*.xspf=38;5;45:', 'GJS_DEBUG_TOPICS': 'JS ERROR;JS LOG', 'XDG_SEAT': 'seat0'}

ivankravets, 2017-09-21:
Fixed in PIO Core+ 0.10.10. Please run `pio update` and restart IDE

inghardware2, 2018-06-13:
I am using Manjaro Deepin and it does not load the list of boards, I have done everything that you have put in your indications and still I can not solve this problem, I have VS Code and I installed the PlatformIO core 3.6.0a2  extension

inghardware2, 2018-06-13:
How do I fix PIO Core+ 0.10.10?

ivankravets, 2018-06-15:
@inghardware2 could you provide an output of `python -c "import os; print os.environ"`?

ivankravets, 2018-06-15:
@inghardware2  https://github.com/platformio/platformio-core/issues/1311

closed 2018-06-15 by ivankravets