platformio/platformio-core #1055: Home: Internal Store Exception

Bruce313, 2017-09-09:
Invariant Violation: Minified React error #31; visit http://facebook.github.io/react/docs/error-decoder.html?invariant=31&args[]=object%20with%20keys%20%7Burl%2C%20name%2C%20email%7D&args[]= for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
    at r (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:11:1349)
    at a (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:121427)
    at a (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:121147)
    at o (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:121557)
    at Object.instantiateChildren (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:520768)
    at h._reconcilerInstantiateChildren (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:518723)
    at h.mountChildren (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:518930)
    at h._createInitialChildren (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:507788)
    at h.mountComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:506002)
    at Object.mountComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:17:30410)

ivankravets, 2017-09-09:
Could you explain in details how to reproduce this issue?

Bruce313, 2017-09-11:
Hi.
I use vscode(1.16.0)  on mac(10.12.6 ).
then I install  platformio plugin to vscode.
error occur when I  Libraries -> Installed -> "click item with 'tiny-http'"
is there a server with port 8010 running in background?

ivankravets, 2017-09-11:
Could you kill all `python/pioplus` processes and restart VSCode? Does it work now?

commit 25da978f (2017-09-21): Add support for "author" field in library.json manifest // Resolve #1055
closed 2017-09-21 by ivankravets