platformio/platformio-core #1058: Home: Internal Store Exception

dikon25, 2017-09-10:
TypeError: Cannot read property 'indexOf' of null
    at Object.o [as basename] (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:11:49023)
    at t.value (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:85404)
    at t.value (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:84226)
    at d._renderValidatedComponentWithoutOwnerOrContext (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:528221)
    at d._renderValidatedComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:528348)
    at d._updateRenderedComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:527652)
    at d._performComponentUpdate (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:527453)
    at d.updateComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:526731)
    at d.receiveComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:525848)
    at Object.receiveComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:17:30804)

ivankravets, 2017-09-11:
What is your OS? Could you reproduce it again?

ivankravets, 2017-09-15:
Duplicate of https://github.com/platformio/platformio-core/issues/1062

closed 2017-09-15 by ivankravets