platformio/platformio-core #1059: Home: Internal Store Exception

chappy1978, 2017-09-10:
Trying to Import an Arduino Project. I have to say I am not a fan off the new homepage setup. The old version was much more useable.
TypeError: Cannot read property 'filter' of null
    at t.value (http://127.0.0.1:8011/b7b2e59db5b772f69810.min.js:31:86699)
    at http://127.0.0.1:8011/b7b2e59db5b772f69810.min.js:31:86920
    at Array.map (native)
    at t.value (http://127.0.0.1:8011/b7b2e59db5b772f69810.min.js:31:86749)
    at http://127.0.0.1:8011/b7b2e59db5b772f69810.min.js:31:86920
    at Array.map (native)
    at t.value (http://127.0.0.1:8011/b7b2e59db5b772f69810.min.js:31:86749)
    at http://127.0.0.1:8011/b7b2e59db5b772f69810.min.js:31:86920
    at Array.map (native)
    at t.value (http://127.0.0.1:8011/b7b2e59db5b772f69810.min.js:31:86749)

ivankravets, 2017-09-11:
> Trying to Import an Arduino Project. I have to say I am not a fan off the new homepage setup. The old version was much more useable.

Could you explain in details what is wrong with new PIO Home? We will improve that as soon as possible.

ivankravets, 2017-09-21:
Could you open PIO IDE Terminal and type
```
pio upgrade
pio update
```
Restart IDE.

Does it work now?

closed 2017-10-17 by ivankravets