platformio/platformio-core #1061: PlatformIO update fails behind proxy

ADeeds, 2017-09-12:
------------------------------------------------------------------

### Configuration

**Operating system**:

OSX 10.11.6
 
**PlatformIO Version** (`platformio --version`):

3.4.1

### Description of problem

the `platformio update` command fails when used behind a proxy, even when $http_proxy & friends are properly configured. I belief this is because the internet_on function in util.py tries to make a direct socket connection and barfs when it hits the proxy

#### Steps to Reproduce

1. Install platformio on a computer that is connected to the internet via a proxy
2. Install some libraries, boards, etc. (`platformio install` doesn't check the internet_on function, so it works as expected)
3. Run `platformio update`. Every package shows "off-line" as the status and never gets updated (unless you uninstall it and reinstall)

### Actual Results

Every package shows as [Off-line]

### Expected Results

[Up-to-date] or whatever it says when packages are out of date (and ideally it would update stuff that was old)

### If problems with PlatformIO Build System:

**Source file to reproduce issue:**
```python
import urllib2
import socket

timeout = 3;
try:
	socket.setdefaulttimeout(timeout)
	socket.socket(socket.AF_INET, socket.SOCK_STREAM).connect(('example.com', 80))
	print('yay')
except:
	print('no')
try:
	response = urllib2.urlopen('http://example.com', timeout=timeout)
	print('yay')
except:
	print('no')
```

### Additional info

Replacing the direct socket stuff with urllib2 as shown in the example above seems to do the right thing on my end. Plus I think it fixes any future issue with ipv6 compatibility :)

ivankravets, 2017-09-16:
Thanks a lot for the report! I don't know why didn't use `requests` directly.

Please re-test http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2017-09-16:
Fixed in https://github.com/platformio/platformio-core/commit/fc7f1c0728954a6d13b048ccdca3da756584474d

closed 2017-09-16 by ivankravets

ADeeds, 2017-09-18:
Works great, thanks!

commit fc7f1c07 (2017-09-16): Fix platforms, packages, and libraries updating behind proxy // Resolve #1061