platformio/platformio-core #1062: Home: Internal Store Exception

jeremybuisson, 2017-09-12:
TypeError: Cannot read property 'indexOf' of null
    at Object.o [as basename] (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:11:49023)
    at t.value (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:85404)
    at t.value (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:84226)
    at d._renderValidatedComponentWithoutOwnerOrContext (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:528221)
    at d._renderValidatedComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:528348)
    at d._updateRenderedComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:527652)
    at d._performComponentUpdate (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:527453)
    at d.updateComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:526731)
    at d.receiveComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:31:525848)
    at Object.receiveComponent (http://127.0.0.1:8010/b7b2e59db5b772f69810.min.js:17:30804)

What I did...
1) Get the Code zipfile for Windows x64, unzip it
2) Launch Code, install PlatformIO extension and its dependencies
3) Wait, and reload
4) Wait, and reload
5) Install Adafruit Circuit Playground library
6) New project, select Adafruit Flora, untick "Use default location" ---> crash

ivankravets, 2017-09-13:
Do you have non-ascii chars in folders/files located in %USER_HOME% directory?

jeremybuisson, 2017-09-13:
%USER_HOME% contains non-ascii chars, yes.

ivankravets, 2017-09-13:
Please run this command in PIO IDE Terminal:
```
python -c "import os; print os.listdir(os.path.expanduser('~'))"
```

Do you see non-ascii chars?

jeremybuisson, 2017-09-13:
Silly question... How do I find PIO IDE Terminal?

ivankravets, 2017-09-13:
http://docs.platformio.org/en/latest/ide/vscode.html#platformio-toolbar

jeremybuisson, 2017-09-13:
Would it be the same if I run the command from the C:\.platformio\... location?
If so: yes, there are non-ascii chars in "Voisinage réseau", "Menu Démarrer", "Modèles", etc.
Sorry for the French words with accents...

I'm really sorry if I look dumb, but, in the bottom-left corner, I see only the PIO Home link.
[image]

ivankravets, 2017-09-21:
Could you open PIO IDE Terminal and type
```
pio upgrade
pio update
```
Restart IDE.

Does it work now?

jeremybuisson, 2017-09-22:
I run the commands in the terminal. Now everything is up-to-date with version 3.5.0a15.

But it does not change anything. Now the stack trace is (same error / different file name):
TypeError: Cannot read property 'indexOf' of null at Object.o [as basename] (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:11:40221) at t.value (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:86316) at t.value (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:85138) at d._renderValidatedComponentWithoutOwnerOrContext (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:519352) at d._renderValidatedComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:519479) at d._updateRenderedComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:518783) at d._performComponentUpdate (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:518584) at d.updateComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:517862) at d.receiveComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:516979) at Object.receiveComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:17:27043)

candidoneto89, 2017-09-22:
Having the same issue. I have folders with non-ascii characters too. Tried what was suggested and still facing this problem.

ivankravets, 2017-11-02:
Thanks for the report!

Please open PIO IDE Terminal and type `pio upgrade`. It should be fixed in 3.5.0b1. Don't forget to restart IDE.

closed 2017-11-02 by ivankravets