platformio/platformio-core #1064: Prevent repeat installation of identical libraries vai git vs library registry

SRGDamia1, 2017-09-12:
What kind of issue is this?
- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**:  Windows 7, Atom 1.19.7x64

**PlatformIO Version** (`platformio --version`): 3.5.0a9

### Description of problem
The same library will be installed twice if it is called in different ways by the pio lib install command or within a library dependency list.  (ie, one lists the github url and another lists the library id number)

#### Steps to Reproduce

1.  pio lib install https://github.com/PaulStoffregen/AltSoftSerial.git
2.  pio lib install AltSoftSerial

### Actual Results
Two nearly identical copies of AltSoftSerial are now installed in .piolibdeps, one in the folder "AltSoftSerial" and one in the folder "AltSoftSerial_ID73."  The version installed from the git repository is tied to the latest git commit, the version installed from the library registry is tied to the version number within the library.json (and the git release number).  Both versions seem to be compiled and then the linker decides which version it wants to use.  If the versions are not identical, there may be compiling issues or just weird results in the final progam.

### Expected Results
The library manager detects that there are two libraries with the same name and asks which version you truly want and removes the other.

### Additional info
It's fairly easy to avoid double-installing libraries when installing them manually, but the same thing happens when the library manager automatically installs dependencies.  So if the library.json of one library lists a git repository as a dependency and another lists the same library by its registry name/number, the library manager will continually reinstall both libraries.  Even if you attempt to uninstall one version manually, it will automatically be re-installed the next build or clean.

ivankravets, 2017-09-13:
PlatformIO allows libraries with the same names. There are a lot of libraries with the same name but different compatibility options.

SRGDamia1, 2017-09-13:
In the case of two libraries with the same name, how do I know/control which is being used?

ivankravets, 2017-09-23:
Do you want to install both libraries for the same project and use them for different environemnts?

SRGDamia1, 2017-09-25:
No, I only want one of them, I would like to uninstall the other.  They just keep reinstalling because of the different ways they're listed as dependencies.

ivankravets, 2017-09-25:
Please remove library from `.piolibdeps` and add 1 record to 
```
[env:...]
lib_deps = https://github.com/PaulStoffregen/AltSoftSerial.git
```

closed 2017-09-25 by ivankravets