platformio/platformio-core #1074: Library Dependencies in library.json not fetched for pio CI

cujomalainey, 2017-09-16:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [x] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: OSX 10.12

**PlatformIO Version** (`platformio --version`): 3.4.1

### Description of problem
Library Dependencies are not fetched when running CI

#### Steps to Reproduce

1. Attempt to run `pio ci` on a library with a dependency section

### Actual Results
Dependency is not fetched and dependencies are not met therefore build fails. Installing the library via pio lib install gets the dependent library.
https://travis-ci.org/cujomalainey/antplus-arduino/jobs/276188370

### Expected Results
Dependencies should be fetched for CI

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:teensy31]
platform = teensy
framework = arduino
board = teensy31

[env:nrf52]
platform = nordicnrf52
framework = arduino
board = nrf52_dk
build_flags = -DNRF52_S132
```

**Source file to reproduce issue:**
https://github.com/cujomalainey/antplus-arduino

### Additional info

ivankravets, 2017-09-16:
Thanks! I'll take a look on that very soon! Temporary solution:
```
platformio ci --lib="." --project-option="lib_deps=ANT" --board=teensy31
```

See http://docs.platformio.org/en/latest/userguide/cmd_ci.html#cmdoption-platformio-ci-o

ivankravets, 2017-11-29:
That is normal behavior. You need manually install the library with all dependencies. Please add 

```
....

install:
    - pip install -U platformio
    - platformio lib -g install file://.
...

closed 2017-11-29 by ivankravets

r41d, 2022-05-17:
Is this `platformio lib install file://.` command mentioned in any documentation? Cause I didn't see it anywhere and this issue here is the only place I made a find.

ivankravets, 2022-05-17:
@r41d I removed my comment. What is your use case? Could you put your dependencies to the `lib_deps`?

See https://docs.platformio.org/en/latest/librarymanager/dependencies.html

r41d, 2022-05-17:
I'm currently in the process of migrating a library (ClusterDuck-Protocol) from dependencies via git submodules to dependencies, this has to be done for both Arduino IDE (`library.properties`) and PlatformIO (`library.json`).
Basically, I'm inside the library folder and want to fetch alle dependencies that are listed in `library.json`

ivankravets, 2022-05-17:
If you plan to use your library later with the PlatformIO project, PlatformIO will resolve all dependencies. You don't need to install them manually. 

Please note that you can "symlink" your library in PlatformIO ( https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_install.html#local-folder ) instead of hard copying.