platformio/platformio-core #1080: Stuck in remote device monitor selection

cujomalainey, 2017-09-23:
### Configuration

**Operating system**: OSX trying to use pio remote to a raspberry pi

**PlatformIO Version** (`platformio --version`): 3.4.1

### Description of problem
Stuck in menu, no amount of Ctrl-C can get me out

#### Steps to Reproduce

1. connect to a remote agent serial port
2. (no platforms were installed on remote agent which I think caused the error)
3. Try and escape from the menu

### Actual Results
[image]

### Expected Results
I should be able to control-c out of the menu

ivankravets, 2017-09-23:
Coul you provide an output from `pio remote device list`?

cujomalainey, 2017-09-23:
I reverted the remote agent back to the state I got it stuck in (ie. I removed all the installed platforms then ran the command)

[image]

ivankravets, 2017-09-23:
Please provide an output from Terminal where you run this `pio remote` command
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

cujomalainey, 2017-09-23:
```
PATH = /Users/curtismalainey/.rvm/gems/ruby-2.2.3/bin:/Users/curtismalainey/.rvm/gems/ruby-2.2.3@global/bin:/Users/curtismalainey/.rvm/rubies/ruby-2.2.3/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/MacGPG2/bin:/Users/curtismalainey/.rvm/gems/ruby-2.2.3/bin:/Users/curtismalainey/.rvm/gems/ruby-2.2.3@global/bin:/Users/curtismalainey/.rvm/rubies/ruby-2.2.3/bin:/usr/local/sbin:/Users/curtismalainey/.rvm/bin

PYTHONPATH =

python version = 2.7.10

pip --version = pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)

pip list = 
aniso8601 (1.2.0)
appdirs (1.4.0)
beautifulsoup4 (4.5.1)
behave (1.2.5)
cffi (1.9.1)
click (6.7)
cryptography (1.5.3)
cycler (0.10.0)
decorator (4.0.10)
dj-database-url (0.4.2)
Django (1.10.5)
django-filter (1.0.1)
djangorestframework (3.5.4)
docutils (0.12)
ecdsa (0.13)
enum (0.4.6)
enum34 (1.1.6)
eyeD3 (0.7.9)
Flask (0.12)
Flask-RESTful (0.3.5)
funcsigs (1.0.2)
future (0.16.0)
gmusicapi (10.1.0)
gpsoauth (0.4.0)
gunicorn (19.6.0)
httplib2 (0.9.2)
idna (2.1)
ipaddress (1.0.18)
itsdangerous (0.24)
Jinja2 (2.9.5)
lockfile (0.12.2)
Markdown (2.6.8)
MarkupSafe (0.23)
matplotlib (1.5.3)
matrixtoolkit (0.6.3)
mbed-cli (1.1.1)
MechanicalSoup (0.6.0)
mock (2.0.0)
mutagen (1.35.1)
ndg-httpsclient (0.4.2)
netifaces (0.10.5)
nrfutil (3.3.0)
numpy (1.11.2)
oauth2client (4.0.0)
olefile (0.44)
packaging (16.8)
parse (1.8.2)
parse-type (0.3.4)
pbr (1.10.0)
pc-ble-driver-py (0.11.1)
piccata (1.0.0)
Pillow (4.0.0)
pip (9.0.1)
powerline-status (2.6)
proboscis (1.2.6.0)
protobuf (3.4.0)
psycopg2 (2.6.2)
pyasn1 (0.1.9)
pyasn1-modules (0.0.8)
PyAudio (0.2.9)
pycairo (1.15.3)
pycparser (2.17)
pycryptodome (3.4.3)
pycryptodomex (3.4.3)
pyFFTW (0.10.4)
pygobject (3.26.0)
pyOpenSSL (16.2.0)
pyparsing (2.1.10)
pyserial (3.4)
pyspinel (1.0.0a1)
python-daemon (2.1.2)
python-dateutil (2.6.0)
pytz (2016.7)
pyxdg (0.25)
requests (2.12.1)
rsa (3.4.2)
scipy (0.18.1)
setuptools (36.5.0)
six (1.11.0)
tqdm (4.15.0)
validictory (1.1.0)
virtualenv (15.1.0)
Werkzeug (0.11.15)
wheel (0.29.0)
whitenoise (3.3.0)
wrapt (1.10.11)
```

Note PIO was not installed using pip, it was installed using `brew`

ivankravets, 2017-09-23:
Please check `pio device list` on maOS side nd RPi. Thanks.

cujomalainey, 2017-09-23:
OSX
[image]

RPI
[image]

cujomalainey, 2017-09-23:
The device list error was triggered because I forgot to run the agent as root.

ivankravets, 2017-09-23:
Did you install https://github.com/platformio/platformio-core/blob/develop/scripts/99-platformio-udev.rules and `sudo usermod -a -G dialout $USER` ?

cujomalainey, 2017-09-23:
I just installed the rules and added the user group, still failing on the default user

ivankravets, 2017-09-25:
Hm... I can't reproduce this issue. See attached picture.

Could you try to install PlatformIO in virtual environment?

```
cd /tmp
virtualenv piotmp
source piotmp/bin/activate
pio remote agent start
```

Does it work now?

[image]

cujomalainey, 2017-09-26:
That worked. Hmm, something is screwy with my install then. It appeared to use the same cache of already installed platform packages rather than reinstalling them into the local venv.

ivankravets, 2017-09-27:
Could I close this issue?

cujomalainey, 2017-09-27:
That doesn't resolve the UI issue where the user is stuck when there is an error. That was the main point of the issue.

ivankravets, 2017-09-27:
I don't see `platformio` in `pip --list` about. How did you install it?

cujomalainey, 2017-09-27:
> Note PIO was not installed using pip, it was installed using `brew`

ivankravets, 2017-09-27:
Please provide an output from RPi
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

cujomalainey, 2017-09-28:
```
PATH= /home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
PYTHONPATH=
python --version = 2.7.9
pip --version = 1.5.6
pip list
argparse (1.2.1)
arrow (0.10.0)
BiblioPixel (2.1.0)
blinker (1.3)
bottle (0.12.13)
chardet (2.3.0)
click (5.1)
colorama (0.3.2)
Flask (0.10.1)
gpiozero (1.4.0)
html5lib (0.999)
itsdangerous (0.24)
Jinja2 (2.7.3)
lockfile (0.12.2)
lxkeymap (0.1)
MarkupSafe (0.23)
mcpi (0.1.1)
ndg-httpsclient (0.3.2)
numpy (1.8.2)
picamera (1.13)
picraft (1.0)
pifacecommon (4.2.1)
pifacedigitalio (3.1.0)
pigpio (1.35)
Pillow (2.6.1)
pip (1.5.6)
platformio (3.4.1)
pyasn1 (0.1.7)
pygame (1.9.2a0)
PyGithub (1.29)
pygobject (3.14.0)
pyinotify (0.9.4)
pyOpenSSL (0.13.1)
pyserial (3.4)
python-apt (0.9.3.12)
python-dateutil (2.6.1)
requests (2.4.3)
RPi.GPIO (0.6.3)
RTIMULib (7.2.1)
semantic-version (2.6.0)
sense-emu (1.0)
sense-hat (2.2.0)
setuptools (5.5.1)
six (1.8.0)
smbus (1.1)
spidev (3.0)
twython (3.1.2)
urllib3 (1.9.1)
Werkzeug (0.9.6)
wheel (0.24.0)
wsgiref (0.1.2)
```

ivankravets, 2017-11-29:
Could you reproduce this issue with the latest PIO Core 3.5.0rc3?

cujomalainey, 2017-12-09:
How do switch the rc release stream?

ivankravets, 2017-12-10:
http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2017-12-15:
Do you have any news?

cujomalainey, 2017-12-15:
Unfortunately I won't be able to test till after Christmas, I am in the middle of moving. I will test this though.

cujomalainey, 2018-01-06:
Cannot reproduce.

ivankravets, 2018-01-06:
Thanks for the report

closed 2018-01-06 by ivankravets