platformio/platformio-core #1082: Home: Could not load boards list

dhoney, 2017-09-23:
PIO Core Call Error: Extra data: line 3 column 1 - line 10 column 63 (char 141703 - 142170)

ivankravets, 2017-09-23:
Please open PlatformIO IDE Terminal and type
```
pio boards --json-output
```

Do you see any errors?

dhoney, 2017-09-24:
I notice some error output at the start followed by a huge json list of boards

```
pio boards --json-output
sh: scl: line 1: syntax error: unexpected end of file
sh: error importing function definition for `BASH_FUNC_scl'
sh: module: line 1: syntax error: unexpected end of file
sh: error importing function definition for `BASH_FUNC_module'
sh: scl: line 1: syntax error: unexpected end of file
sh: error importing function definition for `BASH_FUNC_scl'
sh: module: line 1: syntax error: unexpected end of file
sh: error importing function definition for `BASH_FUNC_module'
[{"vendor": "4DSystems", "rom": 520192, "frameworks": ["arduino"], "url": "http://www.4dsystems.com.au/product/Picadillo_35T/", "fcpu": 80000000, "ram": 131072, "connectivity": null, "platform"
: "microchippic32", "debug": null, "mcu": "32MX795F512L", "id": "picadillo_35t", "name": "4DSystems PICadillo 35T"}, {"vendor": "96Boards", "rom": 524288, "frameworks": ["mbed", "stm32cube"], "
url": "https://developer.mbed.org/platforms/ST-B96B-F446VE/", "fcpu": 168000000, "ram": 131072, "connectivity": ["can"], "platform": "ststm32", "debug": {"tools": {"stlink-v2-1": {"onboard": tr

ue}}}, "mcu": "STM32F446VET6", "id": "b96b_f446ve", "name": "96Boards B96B-F446VE"}, {"vendor": "NXP", "rom": 32768, "frameworks": ["mbed"], "url": "https://developer.mbed.org/handbook/mbed-NXP
-LPC11U24", "fcpu": 48000000, "ram": 8192, "connectivity": null, "platform": "nxplpc", "debug": {"tools": {"cmsis-dap": {"onboard": true}}}, "mcu": "LPC11U24", "id": "lpc11u24_301", "name": "AR
M mbed LPC11U24 (+CAN)"}, {"vendor": "Adafruit", "rom": 28672, "frameworks": ["arduino"], "url": "https://www.adafruit.com/products/2661", "fcpu": 8000000, "ram": 2560, "connectivity": ["blueto
oth"], "platform": "atmelavr", "debug": null, "mcu": "ATMEGA32U4", "id": "bluefruitmicro", "name": "Adafruit Bluefruit Micro"}, {"vendor": "Adafruit", "rom": 262144, "frameworks": ["arduino"],
"url": "https://www.adafruit.com/product/3000", "fcpu": 48000000, "ram": 32768, "connectivity": null, "platform": "atmelsam", "debug": {"tools": {"blackmagic": {}, "cmsis-dap": {}}}, "mcu": "SA
MD21G18A", "id": "adafruit_circuitplayground_m0", "name": "Adafruit Circuit Playground Express"}, {"vendor": "Adafruit", "rom": 1310720, "frameworks": ["arduino", "espidf"], "url": "https://www
.adafruit.com/product/3405", "fcpu": 240000000, "ram": 294912, "connectivity": ["wifi", "bluetooth", "ethernet", "can"], "platform": "espressif32", "debug": null, "mcu": "ESP32", "id": "feather
esp32", "name": "Adafruit ESP32 Feather"}, {"vendor": "Adafruit", "rom": 28672, "frameworks": ["arduino"], "url": "https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/", "fcpu": 80000
00, "ram": 2560, "connectivity": null, "platform": "atmelavr", "debug": null, "mcu": "ATMEGA32U4", "id": "feather32u4", "name": "Adafruit Feather"}, {"vendor": "Adafruit", "rom": 262144, "frame
works": ["arduino"], "url": "https://www.adafruit.com/product/2772", "fcpu": 48000000, "ram": 32768, "connectivity": null, "platform": "atmelsam", "debug": {"tools": {"blackmagic": {}, "cmsis-d
ap": {}}}, "mcu": "SAMD21G18A", "id": "adafruit_feather_m0", "name": "Adafruit Feather M0"}, {"vendor": "Adafruit", "rom": 262144, "frameworks": ["arduino"], "url": "https://www.adafruit.com/pr
oduct/3403", "fcpu": 48000000, "ram": 32768, "connectivity": null, "platform": "atmelsam", "debug": null, "mcu": "SAMD21G18A", "id": "adafruit_feather_m0_express", "name": "Adafruit Feather M0
Express"}, {"vendor": "Adafruit", "rom": 28672, "frameworks": ["arduino"], "url": "http://www.adafruit.com/product/659", "fcpu": 8000000, "ram": 2560, "connectivity": null, "platform": "atmelav
r", "debug": null, "mcu": "ATMEGA32U4", "id": "flora8", "name": "Adafruit Flora"}, {"vendor": "Adafruit", "rom": 8192, "frameworks": ["arduino"], "url": "http://www.adafruit.com/products/1222",
 "fcpu": 8000000, "ram": 512, "connectivity": null, "platform": "atmelavr", "debug": null, "mcu": "ATTINY85", "id": "gemma", "name": "Adafruit Gemma"},...```

ivankravets, 2017-09-25:
What is your OS?

dhoney, 2017-09-25:
Fedora 23

ivankravets, 2017-09-27:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

dhoney, 2017-09-28:
```
$ echo $PATH
/usr/lib64/qt-3.3/bin:/usr/local/heroku/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/home/zero/.local/bin:/home/zero/bin
$ echo $PYHONPATH

$ python --version
Python 2.7.11
$ pip --version
pip 8.1.2 from /usr/lib/python2.7/site-packages (python 2.7)

$ pip list
ansible (2.2.0.0)
argh (0.23.2)
Babel (1.3)
blessings (1.5)
bpython (0.15)
bzr (2.6.0)
BzrTools (2.6.0)
chardet (2.2.1)
click (6.6)
configobj (5.0.6)
curtsies (0.2.11)
decorator (4.0.10)
ecdsa (0.11)
Flask (0.11.1)
greenlet (0.4.9)
gyp (0.1)
httplib2 (0.9.1)
iniparse (0.4)
ino (0.3.6)
ipython (3.2.1)
itsdangerous (0.24)
Jinja2 (2.8)
jsonschema (2.4.0)
kitchen (1.2.1)
MarkupSafe (0.23)
matplotlib (1.4.3)
mercurial (3.5.2)
mistune (0.6)
nose (1.3.7)
numpy (1.9.2)
paramiko (1.15.2)
path.py (5.2)
pathtools (0.1.2)
pexpect (4.0.1)
Pillow (3.0.0)
pip (8.1.2)
ptyprocess (0.5.1)
pyasn1 (0.1.8)
pycrypto (2.6.1)
pycurl (7.19.5.1)
Pygments (2.1.3)
pygobject (3.18.2)
pygpgme (0.3)
pygraphviz (1.3rc2)
pyliblzma (0.5.3)
pyparsing (2.1.5)
pyserial (3.1.1)
PySocks (1.5.6)
python-dateutil (2.4.2)
python-keyczar (0.71rc0)
python-qt (0.50)
pytz (2015.4)
pyxattr (0.5.3)
pyxdg (0.25)
PyYAML (3.11)
pyzmq (14.7.0)
requests (2.10.0)
rpm-python (4.13.0rc1)
setuptools (18.0.1)
simplegeneric (0.8.1)
simpy (2.3.1)
simso (0.8.5)
simsogui (0.8.1)
six (1.9.0)
tornado (4.2.1)
urlgrabber (3.10.1)
urllib3 (1.15.1)
urwid (1.3.0)
vboxapi (1.0)
watchdog (0.8.2)
wcwidth (0.1.6)
Werkzeug (0.11.10)
yum-metadata-parser (1.1.4)
You are using pip version 8.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

ivankravets, 2017-10-19:
Have you resolved this issue?

dhoney, 2017-10-20:
No the issue is still present.

dhoney, 2017-10-20:
I've resolved the issue.

I realized my OS was Fedora 23, 4 versions behind  the current release of fedora 27 and I suspect this unsupported version is what was causing issues with the new platform io update.

I'm on Ubuntu 17.10 now it works just fine.

closed 2017-10-20 by dhoney