platformio/platformio-core #1088: Home: Internal Store Exception

playmean, 2017-09-28:
TypeError: Cannot read property 'indexOf' of null
    at Object.o [as basename] (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:11:40221)
    at t.value (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:86316)
    at t.value (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:85138)
    at d._renderValidatedComponentWithoutOwnerOrContext (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:519352)
    at d._renderValidatedComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:519479)
    at d._updateRenderedComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:518783)
    at d._performComponentUpdate (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:518584)
    at d.updateComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:517862)
    at d.receiveComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:516979)
    at Object.receiveComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:17:27043)

ivankravets, 2017-09-28:
Could you open PIO IDE Terminal and type `pio home`. Does it work now? Do you see any errors?

ivankravets, 2017-10-19:
Duplicate of https://github.com/platformio/platformio-core/issues/1062

closed 2017-10-19 by ivankravets