platformio/platformio-core #1094: Home: Could not load recent projects

carlosmcmarreiros, 2017-09-30:
Server error

ivankravets, 2017-09-30:
Could you open PIO IDE Terminal and type `pio home`. Does it work now? Do you see any errors?

carlosmcmarreiros, 2017-09-30:
still not working i type it and it says pio: command not found

ivankravets, 2017-10-02:
Please click on Terminal button http://docs.platformio.org/en/latest/ide/vscode.html#platformio-toolbar

carlosmcmarreiros, 2017-10-02:
[image]

I dont have botton only home botton. I tried to make new install of vs code and platformio but I have same problem

ivankravets, 2017-10-02:
Try this in Terminal:
```
~/.platformio/penv/bin/pio home
```

carlosmcmarreiros, 2017-10-02:
ok now i get this error 
Unhandled Error
Traceback (most recent call last):
  File "/home/petra/.platformio/packages/pysite-pioplus/txjason/service.py", line 472, in _handle_request
    d = self._call_method(request)
  File "/home/petra/.platformio/packages/pysite-pioplus/twisted/internet/defer.py", line 1445, in unwindGenerator
    return _inlineCallbacks(None, gen, Deferred())
  File "/home/petra/.platformio/packages/pysite-pioplus/twisted/internet/defer.py", line 1299, in _inlineCallbacks
    result = g.send(result)
  File "/home/petra/.platformio/packages/pysite-pioplus/txjason/service.py", line 438, in _call_method
    result = yield defer.maybeDeferred(method, *params)
--- <exception caught here> ---
  File "/home/petra/.platformio/packages/pysite-pioplus/twisted/internet/defer.py", line 150, in maybeDeferred
    result = f(*args, **kw)
  File "cmd_home.py", line 132, in cmd_home.ProjectRPC.get_projects

  File "/home/petra/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/platform.py", line 183, in board_config
    for manifest in self.get_installed_boards():
  File "/home/petra/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/platform.py", line 157, in get_installed_boards
    for manifest in self.get_installed():
  File "/home/petra/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/package.py", line 268, in get_installed
    manifest = self.load_manifest(pkg_dir)
  File "/home/petra/.platformio/penv/local/lib/python2.7/site-packages/platformio/managers/package.py", line 240, in load_manifest
    manifest = util.load_json(manifest_path)
  File "/home/petra/.platformio/penv/local/lib/python2.7/site-packages/platformio/util.py", line 197, in load_json
    "Could not load broken JSON: %s" % file_path)
platformio.exception.PlatformioException: Could not load broken JSON: /home/petra/.platformio/platforms/espressif8266/platform.json

ivankravets, 2017-10-03:
Please share here the content of `/home/petra/.platformio/platforms/espressif8266/platform.json`

carlosmcmarreiros, 2017-10-03:
[platform.json.zip](https://github.com/platformio/platformio-core/files/1352973/platform.json.zip)
i zip it to share it

ivankravets, 2017-10-04:
Please remove`/home/petra/.platformio/platforms/espressif8266/` folder and install dev/platform again.

closed 2017-10-04 by ivankravets

carlosmcmarreiros, 2017-10-04:
thanks very much for your help. keep good working

ivankravets, 2017-10-04:
Thank you too for your report!