platformio/platformio-core #1098: Library dependency resolution wrong for ESP32?

andig, 2017-10-01:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: OSX

**PlatformIO Version** (`platformio --version`): 3.5.0a16

### Description of problem

Compiling for ESP32 pulls in wrong dependecies:

    [env:esp32]
    platform=espressif32_stage
    framework=arduino
    board=featheresp32
    lib_deps =
      1
      19
      31
      54
      64
      ESPAsyncWebServer
      HTTPClient

### Actual Results

Gives

	Collected 32 compatible libraries
	Looking for dependencies...
	Library Dependency Graph
	|-- <ESPAsyncWebServer> v1.1.0
	|   |-- <ESPAsyncTCP> v1.1.0
	|   |-- <FS> v1.0
	|   |-- <WiFi> v1.0
	|   |-- <ArduinoJson> v5.11.1
	|-- <HTTPClient> v1.1
	|   |-- <WiFi> v1.0
	|-- <FS> v1.0
	|-- <ArduinoOTA> v1.0
	|   |-- <WiFi> v1.0
	|   |-- <Update> v1.0
	|-- <SPIFFS> v1.0
	|   |-- <FS> v1.0
	|-- <ArduinoJson> v5.11.1
	|-- <WiFi> v1.0
	|-- <DHT sensor library> v1.3.0
	|   |-- <Adafruit Unified Sensor> v1.0.2
	|-- <OneWire> v2.3.2
	|-- <DallasTemperature> v3.7.7
	|   |-- <OneWire> v2.3.2

### Expected Results

`ESPAsyncWebServer` should really rely on `AsyncTCP` according to https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/espressif32_library.json and not on `ArduinoJson` at all?

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
#env_default = esp8266
env_default = esp32

[env:esp8266]
platform=espressif8266
framework=arduino
board=esp12e
lib_deps =
  1
  19
  31
  54
  64
  ESPAsyncWebServer
build_flags = -Tesp8266.flash.4m1m.ld
upload_port = vzero-edd834.local
#targets = upload

[env:esp32]
platform=espressif32_stage
framework=arduino
board=featheresp32
lib_deps =
  1
  19
  31
  54
  64
  ESPAsyncWebServer
  HTTPClient
#build_flags = -Tesp8266.flash.4m1m.ld
#upload_port = vzero-edd834.local

[platformio]
src_dir=.
```

andig, 2017-10-01:
Update: I'm sometimes seeing the problem from the other direction, i.e. compiling for esp8266, too:

	[Sun Oct  1 17:13:51 2017] Processing esp8266 (platform: espressif8266; lib_deps: Hash, ESPAsyncWebServer, ESP8266mDNS; board: esp12e; framework: arduino)
	---------------------------------------------------------------------------------------------------------------------------------------
	Verbose mode can be enabled via `-v, --verbose` option
	Collected 35 compatible libraries
	Looking for dependencies...
	Library Dependency Graph
	|-- <Hash> v1.0
	|-- <ESPAsyncWebServer> v1.1.0
	|   |-- <ESPAsyncTCP> v1.1.0
	|   |   |-- <ESP8266WiFi> v1.0
	|   |-- <AsyncTCP> v1.0.0
	|   |-- <ESP8266WiFi> v1.0
	|   |-- <Hash> v1.0
	|-- <ESP8266mDNS>
	|   |-- <ESP8266WiFi> v1.0
	|-- <ArduinoOTA> v1.0
	|   |-- <ESP8266WiFi> v1.0
	|   |-- <ESP8266mDNS>
	|   |   |-- <ESP8266WiFi> v1.0
	|-- <DNSServer> v1.1.0
	|   |-- <ESP8266WiFi> v1.0
	|-- <ESP8266WiFi> v1.0

The webserver should not depend on the SyncTCP server here.

Looks as if the LDF is confused by something?

andig, 2017-10-01:
Note: even with `lib_compat_mode` and using https://github.com/me-no-dev/ESPAsyncWebServer instead of the versioned library the esp32 build (but not the esp8266 build) still fails:

	Collected 25 compatible libraries
	Looking for dependencies...
	Library Dependency Graph
	|-- <ESPmDNS> v1.0
	|-- <ESP Async WebServer> v1.1.0
	|   |-- <FS> v1.0
	|   |-- <WiFi> v1.0
	|   |-- <ESPAsyncTCP> v1.1.0
	|-- <ArduinoOTA> v1.0
	|   |-- <Update> v1.0
	|   |-- <WiFi> v1.0
	|   |-- <ESPmDNS> v1.0
	|-- <DNSServer> v1.1.0
	|   |-- <WiFi> v1.0
	Compiling .pioenvs/esp32/src/piotest.o
	Generating partitions .pioenvs/esp32/partitions.bin
	Archiving .pioenvs/esp32/libFrameworkArduinoVariant.a
	Indexing .pioenvs/esp32/libFrameworkArduinoVariant.a
	In file included from src/piotest.cpp:4:0:
	Compiling .pioenvs/esp32/FrameworkArduino/Esp.o
	.piolibdeps/ESP Async WebServer/src/ESPAsyncWebServer.h:33:22: fatal error: AsyncTCP.h: No such file or directory
	compilation terminated.
	*** [.pioenvs/esp32/src/piotest.o] Error 1

The LDF should definitely NOT pull the ESPAsyncTCP dependency in?

andig, 2017-10-01:
Note: I can hack-force it to compile by manually adding required libs and ignoring the wrong ones:

	[platformio]

	[common_env_data]
	lib_compat_mode=2

	[env:esp8266] 
	platform=espressif8266 
	framework=arduino 
	board=esp12e 
	lib_deps =
	  Hash
	  https://github.com/me-no-dev/ESPAsyncWebServer.git
	  ESP8266mDNS
	lib_ignore =
	  AsyncTCP <-- HACK ***

	[env:esp32] 
	platform=espressif32
	framework=arduino
	board=featheresp32
	lib_deps =
	  Hash
	  https://github.com/me-no-dev/ESPAsyncWebServer.git
	  AsyncTCP <-- HACK ***
	  ESPmDNS
	  https://github.com/bbx10/DNSServer_tng
	lib_ignore =
	  ESPAsyncTCP <-- HACK ***

ivankravets, 2017-10-02:
Indeed, it's already fixed in PIO Core 3.5. Just need to rename dev/platform in `feature/stage` branch.

I created related issue https://github.com/platformio/platform-espressif32/issues/48

closed 2017-10-02 by ivankravets

andig, 2017-10-02:
@ivankravets I'm already using PIO Core 3.5a16. Could you explain whats wrong? I'm having exactly the same problem regardless if I use `espressif32` or `espressif32-stage`, also with `lib_compat_mode=2`?

ivankravets, 2017-10-02:
Here is problem
- https://github.com/platformio/platform-espressif32/blob/feature/stage/platform.json#L2

Need to rename it to `espressif32` in "staging" branch. We will do that after PIO Core 3.5 release.

andig, 2017-10-08:
> Need to rename it to espressif32 in "staging" branch. We will do that after PIO Core 3.5 release.

@ivankravets I'm still confused. Even if that screws the esp32 build, why would it have any influence on library resolution for the esp8266 build?

I'm not sure these two are related, but #1111 seems similar to me.