platformio/platformio-core #1102: JLink debug

medovarsky, 2017-10-02:
Hi,

After some time not using JLink, I am unable to run JLink as debugger anymore, using `debug_tool = custom` as documented according to platformio.ini instructions on the [documentation page](http://docs.platformio.org/en/latest/projectconf.html).

On the contrary, I am currently able to debug and upload to STM32L073Z-EVAL board using JLink with extra_script.py as described in issue #1029, using PlatformIO version 3.5.0a13 and VSCode.

My vote for adding JLink as a standard driver.

ivankravets, 2017-10-03:
> After some time not using JLink, I am unable to run JLink as debugger anymore

Do you see any errors? Could you provide your `platformio.ini`?

medovarsky, 2017-10-03:
platformio.ini:
`
[platformio]
env_default = eval_l073z

[env:settings]
platform = ststm32
framework = stm32cube
build_unflags = -Os
build_flags =
    -Og
    -DUSE_FULL_LL_DRIVER
    -DHAL_MODULE_ENABLED
    -DHAL_ADC_MODULE_ENABLED
    -DHAL_COMP_MODULE_ENABLED
    -DHAL_DAC_MODULE_ENABLED
    -DHAL_DMA_MODULE_ENABLED
    -DHAL_FLASH_MODULE_ENABLED
    -DHAL_GPIO_MODULE_ENABLED
    -DHAL_IWDG_MODULE_ENABLED
    -DHAL_LPTIM_MODULE_ENABLED
    -DHAL_PWR_MODULE_ENABLED
    -DHAL_RCC_MODULE_ENABLED
    -DHAL_RTC_MODULE_ENABLED
    -DHAL_TIM_MODULE_ENABLED
    -DHAL_UART_MODULE_ENABLED
    -DHAL_WWDG_MODULE_ENABLED
    -DHAL_CORTEX_MODULE_ENABLED

[env:debug]
build_flags = ${env:settings.build_flags} -DDEBUG

[env:eval_l073z]
platform = ${env:settings.platform}
framework = ${env:settings.framework}
board = eval_l073z
build_flags = ${env:debug.build_flags} -DBOARD_EVAL_L073Z -DSTM32L0XX -DF_CPU=32000000
extra_scripts = extra_script.py
debug_tool = custom
debug_server =
  /usr/bin/JLinkGDBServer
  -singlerun
  -if
  SWD
  -select
  USB
  -port
  2331
  -device
  STM32L073VZ
`

medovarsky, 2017-10-03:
Debug console output:
`
[Mon Oct  2 17:09:25 2017] Processing eval_l073z (extra_scripts: extra_script.py; build_flags: -Og, -DUSE_FULL_LL_DRIVER, -DHAL_MODULE_ENABLED, -DHAL_ADC_MODULE_ENABLED, -DHAL_COMP_MODULE_ENABLED, -DHAL_DAC_MODULE_ENABLED, -DHAL_DMA_MODULE_ENABLED, -DHAL_FLASH_MODULE_ENABLED, -DHAL_GPIO_MODULE_ENABLED, -DHAL_IWDG_MODULE_ENABLED, -DHAL_LPTIM_MODULE_ENABLED, -DHAL_PWR_MODULE_ENABLED, -DHAL_RCC_MODULE_ENABLED, -DHAL_RTC_MODULE_ENABLED, -DHAL_TIM_MODULE_ENABLED, -DHAL_UART_MODULE_ENABLED, -DHAL_WWDG_MODULE_ENABLED, -DHAL_CORTEX_MODULE_ENABLED -DDEBUG -DBOARD_EVAL_L073Z -DSTM32L0XX -DF_CPU=32000000 -DRADIO_RN2483A -DRTC_LSE; debug_tool: custom; platform: ststm32; board: eval_l073z; framework: stm32cube; debug_server: /usr/bin/JLinkGDBServer, -singlerun, -if, SWD, -select, USB, -port, 2331, -device, STM32L073VZ)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 compatible libraries
Looking for dependencies...
Library Dependency Graph
|-- <BSP-STM32L0xx_Nucleo>
Compiling .pioenvs/eval_l073z/src/eval_l073z.o
Compiling .pioenvs/eval_l073z/src/main.o
[redacted]
Compiling .pioenvs/eval_l073z/src/stm32l0xx_it.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_adc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_adc_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_comp.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_comp_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_cortex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_crc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_crc_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_cryp.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_cryp_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_dac.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_dac_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_dma.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_firewall.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_flash.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_flash_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_flash_ramfunc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_gpio.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_i2c.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_i2c_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_i2s.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_irda.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_iwdg.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_lcd.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_lptim.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_pcd.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_pcd_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_pwr.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_pwr_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_rcc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_rcc_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_rng.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_rtc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_rtc_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_smartcard.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_smartcard_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_smbus.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_spi.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_tim.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_tim_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_tsc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_uart.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_uart_ex.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_usart.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_hal_wwdg.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_adc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_comp.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_crc.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_crs.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_dac.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_dma.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_exti.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_gpio.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_i2c.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_lptim.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_lpuart.o
Compiling .pioenvs/eval_l073z/FrameworkHALDriver/Src/stm32l0xx_ll_pwr.o
[... cut ...]
Reading all registers
Starting target CPU...
Reading all registers
Performing single step...
Reading all registers
Performing single step...
Reading all registers
Performing single step...
Reading all registers
Performing single step...
Reading all registers
Starting target CPU...
Reading all registers
Performing single step...
Reading all registers
Performing single step...
Reading all registers
Starting target CPU...
Debugger requested to halt target...
...Target halted (PC = 0x08002A58)
Reading all registers
`

ivankravets, 2017-11-02:
Could we close this issue?

medovarsky, 2017-11-03:
Yes, please close. Still, JLink does not work.

closed 2017-11-04 by ivankravets