platformio/platformio-core #1107: Add support for `inc` folder in a project

ivankravets, 2017-10-05:
* Describe in documentation
* Automatically add `inc` folder to CPPAPTH before building

ivankravets, 2017-12-14:
Resolved via https://github.com/platformio/platformio-core/commit/ca8bc3819fabc9fbeb6078e8042589cbc233c6c7

closed 2017-12-14 by ivankravets

ussserrr, 2018-01-18:
What is purpose of this change? I actually opened my 2-month-ago project and just can't compile it ('not found' error). Only with option `include_dir = inc` compilation was successful.

ivankravets, 2018-01-18:
This is the folder which will be automatically added to CPPPATH. 
Default value is `include`. 

You rename your inc folder to include and remove that extra option

KubaMiszcz, 2018-04-13:
I have project exported from CubeMX, and it generates `Inc` where it keeps .h
after add `include_dir = Inc` below `include_dir = Inc` in `platformio.ini` it still throw error: `src\main.c:41:10: fatal error: main.h: No such file or directory`

for me making a symlink inside root projectfolder: `mklink /d Include Inc`  (inside root) helps

it is better than rename Inc to Includes after every regenerate project in CubeMX

ussserrr, 2018-04-13:
@KubaMiszcz you should add `include_dir` option below dedicated `[platformio]` section of platformio.ini:
```
...

[platformio]
include_dir = Inc
```
See [docs](http://docs.platformio.org/en/latest/projectconf/section_platformio.html#include-dir) for more information.
For me, I preferred to write a small python utility that automate creation and updating PlatformIO projects with CubeMX: [stm32pio](https://github.com/ussserrr/stm32pio). It also handles `Inc` and `Include` cases.

KubaMiszcz, 2018-04-13:
youre right, it works perfectly, I missed this `[platformio]` tag, but it should be more clearly marked in Docs :]