platformio/platformio-core #1111: LDF resolution broken

andig, 2017-10-08:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: OSX sierra

**PlatformIO Version** (`platformio --version`): 3.5.0a16

### Description of problem

Trying to build a dual-build setup for esp8266 and esp32. Using `lib_compat_mode=2` for dependency resolution. Problem is that it's virtually impossible to find a combination that will compile for both platforms, regardless if stage or stable. One aspect of library resolution will always be failing. With `lib_ldf_mode=off` unspecific linker errors. With `lib_ldf_mode=chain`, `MDNS` is not found. Worse, with `lib_ldf_mode=chain+` or even `deep+` `MDNS` is still not found.

#### Steps to Reproduce

1. Include `ESP8266mDNS.h` in `ifdef` for platform
2. Compile
3. Build fails with `MDNS` not found

### Actual Results

Build failure:

	/Users/andig/Documents/htdocs/vzero/vzero.ino: In function 'void start_ota()':
	/Users/andig/Documents/htdocs/vzero/vzero.ino:133:7: error: 'MDNS' was not declared in this scope
	if (MDNS.begin(net_hostname.c_str())) {

### Expected Results

Clean build.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[common_env_data]
lib_compat_mode=2
lib_ldf_mode=deep+

[env:esp8266]
platform=espressif8266
framework=arduino
board=esp12e
lib_deps =
  ArduinoOTA
  Hash
#  ESPAsyncWebServer
  https://github.com/me-no-dev/ESPAsyncWebServer.git
  OneWire
  DHT Sensor Library
  Adafruit Unified Sensor
  DallasTemperature
  ArduinoJson
  ESP8266mDNS
  DNSServer
#  ESPAsyncTCP
  https://github.com/me-no-dev/ESPAsyncTCP.git
```

**Source file to reproduce issue:**
```cpp
#ifdef ESP8266
#include <ESP8266WiFi.h>
#ifdef OTA_SERVER
  #include <ESP8266mDNS.h>
#endif
#endif

void start_ota() {
#ifdef OTA_SERVER
  if (MDNS.begin(net_hostname.c_str())) {
    DEBUG_MSG(CORE, "mDNS responder started at %s.local\n", net_hostname.c_str());
  }
  else {
    DEBUG_MSG(CORE, "error setting up mDNS responder\n");
  }
}
```

### Additional info

andig, 2017-10-19:
ping @ivankravets is the library resolution something you could look into? Full ini file/ code is at https://github.com/andig/vzero/blob/master/platformio.ini

andig, 2017-10-30:
I believe the issue is due to nested pragmas. Same behaviour with flat but multi-condition ifdefs. Converting to flat hierarchy solved it for me.

andig, 2017-11-03:
@ivan apart from playing with the defines- is it correct to put these statements into the common section, i.e. will they be active when doing so:

	[common_env_data]
	lib_compat_mode=2
	lib_ldf_mode=deep+

Looking at https://github.com/platformio/platformio-core/issues/1150 my feeling is that this is not even the right place?

ivankravets, 2017-11-24:
Please read http://docs.platformio.org/en/latest/projectconf/dynamic_variables.html

You need to assign template variable to a real option in [env:...] group

closed 2017-11-24 by ivankravets