platformio/platformio-core #1114: Home: Internal Store Exception

alexandresoaresilva, 2017-10-10:
TypeError: Cannot read property 'filter' of null
    at t.value (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:87611)
    at http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:87297
    at Array.map (native)
    at t.value (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:87152)
    at t.value (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:85224)
    at d._renderValidatedComponentWithoutOwnerOrContext (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:519352)
    at d._renderValidatedComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:519479)
    at d._updateRenderedComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:518783)
    at d._performComponentUpdate (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:518584)
    at d.updateComponent (http://127.0.0.1:8010/993c9d594c3963b6a512.min.js:31:517862)

ivankravets, 2017-10-17:
Please open PlatformIO IDE Terminal and type
```
pio home
```

Now, it should be opened in your browser. Do you see any errors in a console?

ivankravets, 2017-10-19:
Duplicate of https://github.com/platformio/platformio-core/issues/1088

closed 2017-10-19 by ivankravets