platformio/platformio-core #1124: Home: Internal Store Exception

ondrabeni, 2017-10-16:
**# I have this proplem can anybody solve it?** TypeError: Cannot read property 'indexOf' of null
    at Object.o [as basename] (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:11:40274)
    at t.value (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:86403)
    at t.value (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:85225)
    at d._renderValidatedComponentWithoutOwnerOrContext (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:479053)
    at d._renderValidatedComponent (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:479180)
    at d._updateRenderedComponent (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:478484)
    at d._performComponentUpdate (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:478285)
    at d.updateComponent (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:477563)
    at d.receiveComponent (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:25:476680)
    at Object.receiveComponent (http://127.0.0.1:8011/8a915bf8c2b0eb535301.min.js:11:95961)

ivankravets, 2017-10-17:
Please open PlatformIO IDE Terminal and type
```
pio home
```

Now, it should be opened in your browser. Do you see any errors in a console?

ivankravets, 2017-10-19:
Duplicate of https://github.com/platformio/platformio-core/issues/1088

closed 2017-10-19 by ivankravets