platformio/platformio-core #1138: Call dependencies by library id in library.json

SRGDamia1, 2017-10-24:
What kind of issue is this?

- [x] Feature Request

Allow specification of numeric library id for dependencies in library.json.  This should make it clear exactly which registered library is required in cases of libraries with duplicate names.

ie:
"dependencies": [
  {
   "name": "OneWire"
   "id":  #1
  }
]

ivankravets, 2017-11-02:
I don't recommend using IDs in manifest files. Manifest should be readable for people and other library managers. 
If you need to be sure that you depend on a right library, you can extend criteria with `authors`, `version`, `platforms`, `frameworks`.

cr1st1p, 2019-08-25:
@ivankravets I want to use ArduinoJWT library. There are 2 of them. Same author. Different version numbers though.
I can use library ID in platform.ini to load the correct one but not in library.json.
Any suggestion on how to overcome this issue?

Separately:  unfortunately, if I try in platform.ini to load it by version, it still picks the wrong one and then fails to find the asked version.