platformio/platformio-core #1139: Home: Libraries: Could not install library

ddjikic, 2017-10-24:
PIO Core Call Error: Library Storage: /home/damir/.platformio/lib
LibraryManager: Installing id=1766
Downloading...
Unpacking...
EspSpiffsService @ a9f97052d3 has been successfully installed!
Installing dependencies
Looking for EspCoreArchitecture library in registry
Found: http://platformio.org/lib/show/1761/EspCoreArchitecture
LibraryManager: Installing id=1761 @ *

Error: Could not find a version that satisfies the requirement '*' for your system 'linux_x86_64'

ivankravets, 2017-10-30:
* Do you have problems with Internet/Permissions?
* Could you install other libraries?

ddjikic, 2017-10-30:
I can install others without any issue, the internet is working, permissions where good.

ivankravets, 2017-10-30:
Please open PIO IDE Terminal and type `pio update`. Then, restart IDE. Does it work now?

ddjikic, 2017-10-30:
still same error msg

ivankravets, 2017-10-31:
Could you try to run in PIO IDE Terminal
```
pio lib -g install EspCoreArchitecture
```

ddjikic, 2017-10-31:
damir@damir-N76VB:~/Documents/PlatformIO/Projects/171025-001001-nodemcuv2$ pio lib -g install EspCoreArchitecture
Library Storage: /home/damir/.platformio/lib
LibraryManager: Installing id=1761
EspCoreArchitecture @ 3fe2c50406 is already installed

ivankravets, 2017-11-03:
Sorry, I can't reproduce this issue :(

Could you reproduce it again?

closed 2017-11-07 by ivankravets