platformio/platformio-core #1146: Home: Libraries: Stats

Liuren1993, 2017-10-29:
PIO Core Call Error: Error: [API] Invalid response: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.w3.org/TR/html4/loose.dtd">
<html><head>
<title>Proxy error: 502 Read from server failed: Unknown error.</title>
</head><body>
<h1>502 Read from server failed: Unknown error</h1>
<p>The following error occurred while trying to access <strong>http://api.platformio.org/lib/stats</strong>:<br><br>
<strong>502 Read from server failed: Unknown error</strong></p>
<hr>Generated Sun, 29 Oct 2017 04:09:28 GMT by Polipo on <em>FRIDO-20170802P.localdomain:8123</em>.
</body></html>

ivankravets, 2017-10-31:
Could you open http://api.platformio.org/lib/stats ?

ivankravets, 2017-11-02:
@
Please reopen if you still need help

closed 2017-11-02 by ivankravets