platformio/platformio-core #1150: Pio run does not include/print library dependencies from common_env_data

andig, 2017-10-30:
It would be great if `lib_deps` could be placed inside `common_env_data`. Currently that does not seem to be the case. 

`pio run` does neither show a warning about an invalid key in `common_env_data` nor include the libraries.

ivankravets, 2017-11-02:
Could you provide an example with `platformio.ini`?

andig, 2017-11-03:
Looking at http://docs.platformio.org/en/latest/projectconf.html and

    ; You MUST inject these options into [env:] section
    ; using ${common_env_data.***} (see below)

I think this is the expected bevaviour.

	[common_env_data]
	lib_deps =
	    OneWire

Is simply not expected to work?

andig, 2017-11-08:
ping @ivankravets I think we can close this one unless you want to change the `[common_env_data]` behaviour?

ivankravets, 2017-11-08:
I don't understand this issue. Could you explain me?

andig, 2017-11-09:
My problem was putting `lib_deps` into `[common_env_data]` and expecting the libs to be included for multiple environments:

	[common_env_data]
	lib_deps=
	  ESPAsyncWebServer@^1.1
	  OneWire@^2.3
	  DHT Sensor Library@^1.3
	  DallasTemperature@^3.7

	[env:esp8266]
	platform=espressif8266
	framework=arduino
        # no lib_deps here

	[env:esp32]
	platform=espressif32_stage
	framework=arduino
        # no lib_deps here

But I think this is simply wrong since the libs need to be imported into every `env` section:

	[env:esp8266]
	lib_deps=
	  ${common_env_data.lib_deps}

If this understanding is correct we can close, no enhancement needed, just user too stupid...

ivankravets, 2017-11-09:
Sure, you should include template in real environment block.

closed 2017-11-09 by ivankravets