platformio/platformio-core #1153: lib_deps piolibdeps issue creates dependences on all platforms

txinto, 2017-10-31:
What kind of issue is this?

A library dependency and env platforms crossed issue.  Piolibdeps directory is applying lib_deps in wrong compilation environments.

- [c ] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows x64

**PlatformIO Version** (`platformio --version`): 3.5.0a16

### Description of problem

I have a project that has to be compiled on many platforms.  The Arduino Uno is the last platform environment of my platformio.ini file, and it is the only one with the "Arduino Motor Shield" library dependence.  With an empty .piolibdeps everything is compiling without problems, but when I try to launch another compilation, all the other platforms fail (Teensy, ESP32...).  The reason is that the first compilation created a .piolibdeps/ArduinoMotorShieldLibrary_ID28 folder, and this folder is being compiled in all the platforms (not only on Uno).  As a result, the second time you compile, it fails.  

#### Steps to Reproduce

1. Use the given platformio.ini file in a project.
2. Compile it once --> ok.
3. Compile it twice --> error!

### Actual Results

Tries to compile wrong libraries in compilation environment that do not depend on them.

### Expected Results

The compilation environment must only compile the libraries they depend on.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[platformio]
; env_default = esp32doit-devkit-v1
; --> ESP32_DEV
; -DARDUINO=10610 -DARDUINO_ARCH_ESP32 -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPLATFORMIO=30500 -DESP32_DEV -DARDUINO_ESP32_DEV
; env_default = esp32dev
; --> ESP32_DEV
; -DARDUINO=10610 -DARDUINO_ARCH_ESP32 -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPLATFORMIO=30500 -DESP32_DEV -DARDUINO_ESP32_DEV
; env_default = d1_mini
; --> ESP8266_WEMOS_D1MINI
; -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30500 -DESP8266 -DARDUINO_ARCH_ESP8266 -DESP8266_WEMOS_D1MINI
; env_default = nodemcuv2
; --> ARDUINO_ESP8266_NODEMCU
; -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30500 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU
; env_default = teensy31
; --> TEENSY31
; -DF_CPU=72000000L -DLAYOUT_US_ENGLISH -DPLATFORMIO=30500 -D__MK20DX256__ -DTEENSY31 -DUSB_SERIAL -DARDUINO=10610 -DTEENSYDUINO=136
; env_default = uno
; --> ARDUINO_AVR_UNO
; -DF_CPU=16000000L -DPLATFORMIO=30500 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10619
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
  PubSubClient
  Adafruit NeoPixel

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
  PubSubClient
  Adafruit NeoPixel

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
  PubSubClient
  Adafruit NeoPixel

[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
lib_deps =
  PubSubClient
  Adafruit NeoPixel

[env:teensy31]
platform = teensy
framework = arduino
board = teensy31
lib_deps =
  Adafruit NeoPixel

[env:uno]
platform = atmelavr
board = uno
framework = arduino
lib_deps =
  Adafruit Motor Shield library
  Adafruit NeoPixel

```

universam1, 2017-11-03:
exact same issue here with `PlatformIO, version 3.5.0a17`

ivankravets, 2017-11-24:
It seems that you have IFDEF guard for different dev/Platforms. Please take a look at `chain+` of http://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

closed 2017-11-24 by ivankravets

euphi, 2017-12-08:
How do I handle this situation for libraries where I try to have a CI integration?

Please see my question on community.platformio.org: https://community.platformio.org/t/set-lib-ldf-mode-for-libraries-library-json-only/3044