platformio/platformio-core #1154: Wrong includePath when indexing intellisense

guettomusick, 2017-10-31:
When opening visual studio code on a Mac OS High Sierra, autoindexing for intellisense takes palce, and it doesn't include the include path for the std library of C++ under esp8266 platform.

If I manually include this include paths,

"~/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2"
"~.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/xtensa-lx106-elf"

the problem is solved, but when I restart visual studio code, the indexing takes place and those include paths are missing again.

closed 2017-11-01 by ivankravets

ivankravets, 2017-11-01:
Thanks a lot for the report!

Please open PIO IDE Terminal and run 
```
pio upgrade
```

Restart IDE.

guettomusick, 2017-11-01:
Thanks! It work perfectly.

I don't know if it has something to do with platformio or visual studio
code, but the editor is interpreting all files as C instead of C++, for .h
and .cpp files.

Thanks

Fer

ivankravets, 2017-11-01:
I think you can configure that in VSCode. Click on "CPP" in the bottom right corner.

guettomusick, 2017-11-01:
Yes, but previous to some updates of platformio and visual studio code it
auto detects to C instead of C++ as previous. I will try to recreate the
project to check if that solves the problem.

Thabks

commit 86de58b9 (2017-11-01): Fix missing toolchain include paths for project generator // Resolve #1154