platformio/platformio-core #1157: Home: Could not update platform

kakegu, 2017-11-01:
PIO Core Call Error: Platform C:\Users\ThingTrust\.platformio\platforms\espressif8266
--------
Updating espressif8266                   @ 1.4.0          [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating framework-esp8266-rtos-sdk      @ 1.5.0-beta.3   [1.5.0-beta.4]

Uninstalling framework-esp8266-rtos-sdk @ 1.5.0-beta.3: 	[OK]
PackageManager: Installing framework-esp8266-rtos-sdk @ 1.5.0-beta.4

Downloading
Downloading
Unpacking
Unpacking
Warning! Package Mirror: Package version 1.5.0-beta.3 doesn't satisfy requirements 1.5.0-beta.4
Looking for other mirror...

Error: Could not install 'framework-esp8266-rtos-sdk' with version requirements '1.5.0-beta.4' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

ivankravets, 2017-11-03:
@yanbe could you fix a version in https://github.com/yanbe/framework-esp8266-rtos-sdk/raw/master/ESP8266_RTOS_SDK-1.5.0-beta.4.tar.gz and reupload binary?

Thanks!

yanbe, 2017-11-04:
Sorry for inconvenience. The reason of updating framework-esp8266-rtos-sdk failure was because I've forgot updating version string in package.json as well as updating package binary at commit https://github.com/yanbe/framework-esp8266-rtos-sdk/commit/4f59c8300361f870bcaa0a0f9a6358da4cdc6ea9 .

I have just fixed here https://github.com/yanbe/framework-esp8266-rtos-sdk/commit/77be8896d74eaca30c6359bfd86c5a123288076b

and I confirmed that `pio update` , `pio run -t upload`  and `pio run -t monitor` can run successfully.

Thanks.

ivankravets, 2017-11-04:
@yanbe thank you very much! 👍 🍺

yanbe, 2017-11-07:
Oh sorry.
I fixed package.json packaging problem and released binary as `1.5.0-beta.5`.
https://github.com/yanbe/framework-esp8266-rtos-sdk/compare/77be8896d...b334a879

(I thought version string should be re-numbered once contents in archive file is modified in semver version system,)

After that I confirmed `pio run` runs correctly with updating `framework-esp8266-rtos-sdk` package. Also I confirmed  `pio update` shows correct version string.
```
$ pio update
(snip)
Platform Espressif 8266
--------
Updating espressif8266                   @ a9fa96d        [Up-to-date]
Updating framework-esp8266-rtos-sdk      @ 1.5.0-beta.5   [Up-to-date]
```
Thanks.

ivankravets, 2017-11-07:
@yanbe Thanks! ;)

closed 2017-11-07 by ivankravets