platformio/platformio-core #1159: Home: Could not initialize project

mhpardo, 2017-11-02:
Server error

ivankravets, 2017-11-02:
Thanks for the report!

Please open PIO IDE Terminal and type `pio upgrade`. It should be fixed in 3.5.0b1. Don't forget to restart IDE.

closed 2017-11-02 by ivankravets

mhpardo, 2017-11-03:
Hello Dear Ivan Kravets,

   thank you for the fast reply about my error report. I'm really new on
Atom Text Editor and Platform IO IDE.

   I executed the suggested command on PlatformIO Terminal and the
execution was sucessful. But, after that, on restarting the IDE the
PlatformIO tab isn't starting anymore. The loading circle icon still
running and the user interface is not loaded.

   If you could give any other instructions to solve this problem, I will
be very grateful.

   Thank you, sir for all help.

Regards,
Mário Henrique de Souza Pardo

ivankravets, 2017-11-03:
Fixed, please restart IDE.

mhpardo, 2017-11-03:
Yes, it's working normally again!

 Thank you so much, sir.

Regards,
Mário Henrique de Souza Pardo