platformio/platformio-core #1164: Home: Could not load boards list

mogplus88, 2017-11-04:
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 52 (char 142001 - 142101)
reinstalled Platformio core (as pio command not found). did pio upgrade, but same error. It is also complaining about the "nospoof on" command and "spoofalert on" command. But it got past those and told me my system was up to date.

ivankravets, 2017-11-04:
What is your OS? Locale?

ivankravets, 2017-11-07:
Please reopen if you still need help

mogplus8, 2017-11-08:
Hi Ivan,
Thanks for the quick response. Unfortunately I'm nowhere near my computer for this week, I won't be back home until Monday so I can't try it until then. I'll try it again then and let you know how I go. 
I thought I'd answered your question earlier but maybe not, as it doesn't look like it's here! OS is PCLinuxOS (KDE desktop), local is Australia (Melbourne).
Thanks,
Ian

ivankravets, 2017-11-26:
@romankot  do you have the same lines https://github.com/platformio/platformio-core/issues/1164#issue-271169305 ?

closed 2017-12-01 by ivankravets