platformio/platformio-core #1170: espressif8266.git#feature/stage breaks LDF resolution

andig, 2017-11-06:
Not sure this is releated to the platform (and should be reported there) or to the LDF, but here we go.

Consider this`ini`:

    [platformio]

    [common_env_data]
    lib_deps=
      ESPAsyncWebServer@^1.1
      OneWire@^2.3
      DHT Sensor Library@^1.3
      DallasTemperature@^3.7
      ArduinoJson@^5.1

    [env:esp8266]
    #platform=espressif8266
    platform=https://github.com/platformio/platform-espressif8266.git#feature/stage
    board=esp12e
    framework=arduino
    lib_compat_mode=2
    lib_ldf_mode=deep
    build_flags=-Tesp8266.flash.4m1m.ld
    upload_port=vzero-edd834.local
    lib_deps=
      ${common_env_data.lib_deps}
      https://github.com/me-no-dev/ESPAsyncTCP.git
    lib_ignore=
      AsyncTCP

With stage platform the build fails since `ESPAsyncWebServer` is not in the dependency graph and not included:

	# platform=https://github.com/platformio/platform-espressif8266.git#feature/stage

	❯ rm -rf .piolibdeps && pio run -e esp8266
	[Mon Nov  6 10:20:58 2017] Processing esp8266 (build_flags: -Tesp8266.flash.4m1m.ld; upload_port: vzero-edd834.local; lib_deps: ESPAsyncWebServer@^1.1, OneWire@^2.3, DHT Sensor Library@^1.3, DallasTemperature@^3.7, ArduinoJson@^5.1, https://github.com/me-no-dev/ESPAsyncTCP.git; lib_ignore: AsyncTCP; platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; board: esp12e; framework: arduino; lib_ldf_mode: deep; lib_compat_mode: 2)
	---------------------------------------------------------------------------------------------------------------------------------------
	LibraryManager: Installing id=306 @ ^1.1
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=305
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=1 @ ^2.3
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=19 @ ^1.3
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=31
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=54 @ ^3.7
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=64 @ ^5.1
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing ESPAsyncTCP
	git version 2.13.6 (Apple Git-96)
	Cloning into '/Users/andig/Documents/htdocs/vzero/.piolibdeps/_tmp_installing-CuvsaM-package'...
	remote: Counting objects: 30, done.
	remote: Compressing objects: 100% (26/26), done.
	remote: Total 30 (delta 3), reused 17 (delta 0), pack-reused 0
	Unpacking objects: 100% (30/30), done.
	Verbose mode can be enabled via `-v, --verbose` option
	Converting vzero.ino
	Collected 33 compatible libraries
	Looking for dependencies...
	Library Dependency Graph
	|-- <OneWire> v2.3.2
	|-- <DallasTemperature> v3.7.7
	|   |-- <OneWire> v2.3.2
	|-- <ArduinoJson> v5.11.2
	|-- <ESP8266HTTPClient> v1.1
	|   |-- <ESP8266WiFi> v1.0
	|-- <ESP8266WiFi> v1.0
	|-- <ESP8266mDNS>
	|   |-- <ESP8266WiFi> v1.0
	|-- <DHT sensor library> v1.3.0
	|   |-- <Adafruit Unified Sensor> v1.0.2
	|-- <ArduinoOTA> v1.0
	|   |-- <ESP8266mDNS>
	|   |   |-- <ESP8266WiFi> v1.0
	|   |-- <ESP8266WiFi> v1.0
	|-- <DNSServer> v1.1.0
	|   |-- <ESP8266WiFi> v1.0
	Compiling .pioenvs/esp8266/src/config.o
	Compiling .pioenvs/esp8266/src/plugins/AnalogPlugin.o
	Compiling .pioenvs/esp8266/src/plugins/DHTPlugin.o
	Compiling .pioenvs/esp8266/src/plugins/OneWirePlugin.o
	Compiling .pioenvs/esp8266/src/plugins/Plugin.o
	Compiling .pioenvs/esp8266/src/plugins/S0Plugin.o
	Compiling .pioenvs/esp8266/src/plugins/WifiPlugin.o
	Compiling .pioenvs/esp8266/src/urlfunctions.o
	Compiling .pioenvs/esp8266/src/vzero.ino.o
	Compiling .pioenvs/esp8266/src/webserver.o
	src/webserver.cpp:8:31: fatal error: ESPAsyncWebServer.h: No such file or directory
	#include <ESPAsyncWebServer.h>
	^
	compilation terminated.
	*** [.pioenvs/esp8266/src/webserver.o] Error 1

With regular version this works fine:

	# espressif8266

	❯ rm -rf .piolibdeps && pio run -e esp8266
	[Mon Nov  6 10:19:27 2017] Processing esp8266 (build_flags: -Tesp8266.flash.4m1m.ld; upload_port: vzero-edd834.local; lib_deps: ESPAsyncWebServer@^1.1, OneWire@^2.3, DHT Sensor Library@^1.3, DallasTemperature@^3.7, ArduinoJson@^5.1, https://github.com/me-no-dev/ESPAsyncTCP.git; lib_ignore: AsyncTCP; platform: espressif8266; board: esp12e; framework: arduino; lib_ldf_mode: deep; lib_compat_mode: 2)
	---------------------------------------------------------------------------------------------------------------------------------------
	LibraryManager: Installing id=306 @ ^1.1
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=305
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=1 @ ^2.3
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=19 @ ^1.3
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=31
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=54 @ ^3.7
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing id=64 @ ^5.1
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	LibraryManager: Installing ESPAsyncTCP
	git version 2.13.6 (Apple Git-96)
	Cloning into '/Users/andig/Documents/htdocs/vzero/.piolibdeps/_tmp_installing-5HCFaM-package'...
	remote: Counting objects: 30, done.
	remote: Compressing objects: 100% (26/26), done.
	remote: Total 30 (delta 3), reused 17 (delta 0), pack-reused 0
	Unpacking objects: 100% (30/30), done.
	Verbose mode can be enabled via `-v, --verbose` option
	Converting vzero.ino
	Collected 36 compatible libraries
	Looking for dependencies...
	Library Dependency Graph
	|-- <ESPAsyncWebServer> v1.1.0
	|   |-- <ESPAsyncTCP> v1.1.1
	|   |   |-- <ESP8266WiFi> v1.0
	|   |-- <ArduinoJson> v5.11.2
	|   |-- <Hash> v1.0
	|   |-- <ESP8266WiFi> v1.0
	|-- <OneWire> v2.3.2
	|-- <DallasTemperature> v3.7.7
	|   |-- <OneWire> v2.3.2
	|-- <ArduinoJson> v5.11.2
	|-- <ArduinoOTA> v1.0
	|   |-- <ESP8266WiFi> v1.0
	|   |-- <ESP8266mDNS>
	|   |   |-- <ESP8266WiFi> v1.0
	|-- <DHT sensor library> v1.3.0
	|   |-- <Adafruit Unified Sensor> v1.0.2
	|-- <ESP8266WiFi> v1.0
	|-- <ESP8266HTTPClient> v1.1
	|   |-- <ESP8266WiFi> v1.0
	|-- <ESP8266mDNS>
	|   |-- <ESP8266WiFi> v1.0
	|-- <DNSServer> v1.1.0
	|   |-- <ESP8266WiFi> v1.0
	Compiling .pioenvs/esp8266/src/vzero.ino.o
	Compiling .pioenvs/esp8266/src/webserver.o
	Compiling .pioenvs/esp8266/lib/ESPAsyncTCP/AsyncPrinter.o
	Compiling .pioenvs/esp8266/lib/ESPAsyncTCP/ESPAsyncTCP.o
	Compiling .pioenvs/esp8266/lib/ESPAsyncTCP/ESPAsyncTCPbuffer.o
	...

Cache folder has been cleaned in both cases to reduce side effects.

ivankravets, 2017-11-06:
Thanks a lot for your last reports. I understand where is the problem hidden and will try to resolve soon.

andig, 2017-11-07:
> Thanks a lot for your last reports. I understand where is the problem hidden and will try to resolve soon.

Great, I didn't see any pattern. Standing by for testing...

ivankravets, 2017-11-07:
Please run `pio upgrade`. Should be fixed in 3.5.0b3

closed 2017-11-07 by ivankravets

andig, 2017-11-08:
@ivankravets unfortunately still not working for stage version:

	❯ pio upgrade
	Please wait while upgrading PlatformIO...
	PlatformIO has been successfully upgraded to 3.5.0b3!

	❯ rm -rf .piolibdeps && pio run -e esp8266 -vv
	[Wed Nov  8 09:31:27 2017] Processing esp8266 (build_flags: -Tesp8266.flash.4m1m.ld; upload_port: vzero-edd834.local; lib_deps: ESPAsyncWebServer@^1.1, OneWire@^2.3, DHT Sensor Library@^1.3, DallasTemperature@^3.7, ArduinoJson@^5.1, ESPAsyncTCP@^1.1.1; lib_ignore: AsyncTCP; platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; board: esp12e; framework: arduino; lib_ldf_mode: deep; lib_compat_mode: 2)
	---------------------------------------------------------------------------------------------------------------------------------------
	Library Storage: /Users/andig/Documents/htdocs/vzero/.piolibdeps
	Looking for ESPAsyncWebServer library in registry
	Found: http://platformio.org/lib/show/306/ESPAsyncWebServer
	LibraryManager: Installing id=306 @ ^1.1
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	ESPAsyncWebServer @ 1.1.0 has been successfully installed!
	Installing dependencies
	Looking for ESPAsyncTCP library in registry
	Found: http://platformio.org/lib/show/305/ESPAsyncTCP
	LibraryManager: Installing id=305
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	ESPAsyncTCP @ 1.1.1 has been successfully installed!
	Looking for OneWire library in registry
	Found: http://platformio.org/lib/show/1/OneWire
	LibraryManager: Installing id=1 @ ^2.3
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	OneWire @ 2.3.2 has been successfully installed!
	Looking for DHT Sensor Library library in registry
	Found: http://platformio.org/lib/show/19/DHT sensor library
	LibraryManager: Installing id=19 @ ^1.3
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	DHT sensor library @ 1.3.0 has been successfully installed!
	Installing dependencies
	Looking for Adafruit Unified Sensor library in registry
	Found: http://platformio.org/lib/show/31/Adafruit Unified Sensor
	LibraryManager: Installing id=31
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	Adafruit Unified Sensor @ 1.0.2 has been successfully installed!
	Looking for DallasTemperature library in registry
	Found: http://platformio.org/lib/show/54/DallasTemperature
	LibraryManager: Installing id=54 @ ^3.7
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	DallasTemperature @ 3.7.7 has been successfully installed!
	Installing dependencies
	Looking for OneWire library in registry
	Found: http://platformio.org/lib/show/1/OneWire
	LibraryManager: Installing id=1
	OneWire @ 2.3.2 is already installed
	Looking for ArduinoJson library in registry
	Found: http://platformio.org/lib/show/64/ArduinoJson
	LibraryManager: Installing id=64 @ ^5.1
	Downloading  [####################################]  100%
	Unpacking  [####################################]  100%
	ArduinoJson @ 5.11.2 has been successfully installed!
	LibraryManager: Installing id=305 @ ^1.1.1
	ESPAsyncTCP @ 1.1.1 is already installed
	xtensa-lx106-elf-g++ -o "/Users/andig/Documents/htdocs/vzero/src/vzero.ino.cpp" -x c++ -fpreprocessed -dD -E "/var/folders/w5/_c0nb6n90fn96tzw04dtc6240000gn/T/tmpY3ctv7"
	Platform incompatible library /Users/andig/Documents/htdocs/vzero/.piolibdeps/DHT sensor library_ID19
	Platform incompatible library /Users/andig/Documents/htdocs/vzero/.piolibdeps/ESPAsyncTCP_ID305
	Platform incompatible library /Users/andig/.platformio/lib/AsyncMqttClient_ID346
	Platform incompatible library /Users/andig/.platformio/lib/ESPAsyncTCP_ID305
	Platform incompatible library /Users/andig/.platformio/lib/ESPAsyncWebServer_ID306
	Platform incompatible library /Users/andig/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/GDBStub
	Collected 34 compatible libraries
	More details about "Library Compatibility Mode": http://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
[... 43 lines cut ...]
	Looking for DallasTemperature library in registry
	Found: http://platformio.org/lib/show/54/DallasTemperature
	LibraryManager: Installing id=54 @ ^3.7
	Unpacking  [####################################]  100%
	DallasTemperature @ 3.7.7 has been successfully installed!
	Installing dependencies
	Looking for OneWire library in registry
	Found: http://platformio.org/lib/show/1/OneWire
	LibraryManager: Installing id=1
	OneWire @ 2.3.2 is already installed
	Looking for ArduinoJson library in registry
	Found: http://platformio.org/lib/show/64/ArduinoJson
	LibraryManager: Installing id=64 @ ^5.1
	Unpacking  [####################################]  100%
	ArduinoJson @ 5.11.2 has been successfully installed!
	LibraryManager: Installing id=305 @ ^1.1.1
	ESPAsyncTCP @ 1.1.1 is already installed
	xtensa-lx106-elf-g++ -o "/Users/andig/Documents/htdocs/vzero/src/vzero.ino.cpp" -x c++ -fpreprocessed -dD -E "/var/folders/w5/_c0nb6n90fn96tzw04dtc6240000gn/T/tmpkOhOUC"
	Platform incompatible library /Users/andig/Documents/htdocs/vzero/.piolibdeps/DHT sensor library_ID19
	Platform incompatible library /Users/andig/Documents/htdocs/vzero/.piolibdeps/ESPAsyncTCP_ID305
	Platform incompatible library /Users/andig/.platformio/packages/framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910/libraries/GDBStub
	More details about "Library Compatibility Mode": http://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
	Collected 29 compatible libraries
	Looking for dependencies...
	Error: Could not find `ESPAsyncTCP` dependency for `ESPAsyncWebServer` library

andig, 2017-11-08:
Update: cleaning `/Users/andig/.platformio/packages` does not help. 

Interesting is that `/Users/andig/Documents/htdocs/vzero/.piolibdeps/ESPAsyncTCP_ID305` has just been downloaded and is then found to be incompatible?

andig, 2017-11-08:
Update: looking at https://github.com/platformio/platform-espressif8266/blob/feature/stage/platform.json#L2 I see that the platform is `espressif8266_stage` but `ESPAsyncTCP` supports `espressif8266` only in https://github.com/me-no-dev/ESPAsyncTCP/blob/master/library.json#L18.

If this is a bug in the library definition- why does it only appear so inconsistently?

ivankravets, 2017-11-08:
Could you try  https://community.platformio.org/t/esp8266-arduino-krack-fix-compile-using-platformio/2747/17?u=ivankravets ?

```
[env:...]
platform = https://github.com/platformio/platform-espressif8266.git#feature/2.4.0-rc2
...
```

See https://github.com/platformio/platform-espressif8266/tree/feature/2.4.0-rc2

andig, 2017-11-08:
Should work because of the renamed platform? Looks like a good idea but fails after installing libraries:

    Error: Traceback (most recent call last):
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
        cli(None, None, None)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
        return self.main(*args, **kwargs)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
        rv = self.invoke(ctx)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
        return _process_result(sub_ctx.command.invoke(sub_ctx))
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
        return ctx.invoke(self.callback, **ctx.params)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
        return callback(*args, **kwargs)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
        return f(get_current_context(), *args, **kwargs)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 109, in cli
        result = (envname, ep.process())
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 193, in process
        result = self._run()
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 287, in _run
        p = PlatformFactory.newPlatform(self.options['platform'])
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 235, in newPlatform
        cls.get_clsname(name))
    AttributeError: 'module' object has no attribute 'Espressif8266Platform'

ivankravets, 2017-11-08:
Fixed. Please run `pio update`

andig, 2017-11-09:
> Fixed

Confirmed, thank you for all your efforts!

commit 4c170b79 (2017-11-07): Ensure real dev/platform name before build process // Resolve #1170