platformio/platformio-core #1173: pio lib register should allow additional attributes

andig, 2017-11-08:
Coming from https://github.com/platformio/platformio-core/issues/1170 I've noticed that https://github.com/adafruit/DHT-sensor-library does not support `espressif8266` as platform, only arduino. 

For multi-platform pio builds that require ldf_compat_mode=2 that will be a problem (`Platform incompatible library`).

Registering a "foreign" library is already possible using `pio lib register` according to http://docs.platformio.org/en/latest/userguide/lib/cmd_register.html. 

It would be helpful if there was a mechanism to specify additional properties for these foreign libraries like additional supported platforms.

ivankravets, 2017-11-08:
It should help https://github.com/platformio/platformio-core/issues/1170#issuecomment-342785182

andig, 2017-11-08:
Should work because of the renamed platform? Looks like a good idea but fails after installing libraries:

    Error: Traceback (most recent call last):
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
        cli(None, None, None)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
        return self.main(*args, **kwargs)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
        rv = self.invoke(ctx)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
        return _process_result(sub_ctx.command.invoke(sub_ctx))
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
        return ctx.invoke(self.callback, **ctx.params)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
        return callback(*args, **kwargs)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
        return f(get_current_context(), *args, **kwargs)
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 109, in cli
        result = (envname, ep.process())
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 193, in process
        result = self._run()
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 287, in _run
        p = PlatformFactory.newPlatform(self.options['platform'])
      File "/Users/andig/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 235, in newPlatform
        cls.get_clsname(name))
    AttributeError: 'module' object has no attribute 'Espressif8266Platform'

ivankravets, 2017-11-08:
We will remove `_stage` prefix for eps8266 dev/platform soon. It depends on PIO Core 3.5

andig, 2017-11-09:
@ivankravets even when `_stage` is dropped, DHT still doesn't support e.g. `esp8266` as you can reproduce when setting `lib_compat_mode=2`. 

The problem with multi-platform builds is that I need the `lib_compat_mode=2` e.g. for `ESPAsyncWebserver` and its 2 different platform dependencies. When setting this `DHT` is ignored.

I think we still need some kind of mechanism to provide additional compatibility information for non-pio libraries. Maybe user can file issues against the library registry?

andig, 2017-11-09:
Here's an example:

    [platformio]

    [common_env_data]
    lib_deps=
      ESPAsyncWebServer@^1.1
    # Use git version for _stage compatibility (https://github.com/me-no-dev/ESPAsyncWebServer/issues/281)
    #  https://github.com/me-no-dev/ESPAsyncWebServer.git
      OneWire@^2.3
      DHT Sensor Library@^1.3
      DallasTemperature@^3.7
      ArduinoJson@^5.1

    [env:esp8266]
    platform=https://github.com/platformio/platform-espressif8266.git#feature/2.4.0-rc2
    board=esp12e
    framework=arduino
    # lib_compat_mode=1 allows non-git versions of DHT and ESPAsyncWebServer@^1.1
    lib_compat_mode=1
    lib_ldf_mode=deep
    lib_deps=
      ${common_env_data.lib_deps}
      ESPAsyncTCP@^1.1.2
    lib_ignore=
      AsyncTCP

    [env:esp32]
    # use stage version for added SPIFFS support
    platform=espressif32_stage
    board=featheresp32
    framework=arduino
    # lib_compat_mode=2 to avoid including ESPAsyncTCP
    lib_compat_mode=2
    lib_ldf_mode=deep
    lib_deps=
      ${common_env_data.lib_deps}
      https://github.com/bbx10/DNSServer_tng
      AsyncTCP@^1.0
    #  https://github.com/me-no-dev/AsyncTCP.git
    lib_ignore=
      ESPAsyncTCP

The `esp32` needs `lib_compat_mode=2` or it will include `ESPAsyncTCP` which breaks the build.

Unfortunately, with `lib_compat_mode=2` it does no longer include `DHT` (this has changed during the 3.5 pre release versions):

    Library Dependency Graph
    |-- <OneWire> v2.3.2
    |-- <DallasTemperature> v3.7.7
    |   |-- <OneWire> v2.3.2
    |-- <ArduinoJson> v5.11.2
    |-- <AsyncTCP> v1.0.0
    |-- <WiFi> v1.0
    |-- <HTTPClient> v1.1
    |   |-- <WiFiClientSecure> v1.0
    |   |   |-- <WiFi> v1.0
    |   |-- <WiFi> v1.0
    |-- <ArduinoOTA> v1.0
    |   |-- <WiFi> v1.0
    |   |-- <Update> v1.0
    |   |-- <ESPmDNS> v1.0
    |-- <ESPmDNS> v1.0
    |-- <FS> v1.0
    |-- <DNSServer> v1.1.0
    |   |-- <WiFi> v1.0
    |-- <SPIFFS> v1.0
    |   |-- <FS> v1.0

Again, build breaks due to missing `DHT`

ivankravets, 2017-11-25:
Done!

closed 2017-11-25 by ivankravets