platformio/platformio-core #1182: Unable to build in Visual Studio Code, only "pio run" works

tekk, 2017-11-12:
### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): 3.5.0b3

### Description of problem

I am unable to build project via check mark icon in Visual Studio Code. It only works when I click on terminal icon (PIO Terminal) and execute `pio run`.

#### Steps to Reproduce

1. Installed VS Code
2. Installed C++ extension
3. Installed PlatformIO

### Actual Results

```
> Executing task: C:\Python27\Scripts\platformio.exe run <

*************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- donate to keep PlatformIO alive! > http://platformio.org/donate
*************************************************************************************************************************************

Please wait while upgrading PlatformIO ...

Platform atmelavr
--------
Updating toolchain-atmelavr package:
Versions: Current=3, Latest=3    [Up-to-date]
Updating tool-avrdude package:
Versions: Current=4, Latest=4    [Up-to-date]
Updating framework-arduinoavr package:
Versions: Current=32, Latest=32          [Up-to-date]

Platform espressif
--------
Updating toolchain-xtensa package:
Versions: Current=2, Latest=2    [Up-to-date]
Updating tool-esptool package:
Versions: Current=7, Latest=7    [Up-to-date]
Updating tool-mkspiffs package:
Versions: Current=1, Latest=1    [Up-to-date]
Updating framework-arduinoespressif package:
Versions: Current=14, Latest=14          [Up-to-date]
Updating ldscripts package:
Versions: Current=23, Latest=23          [UError: Traceback (most recent call last):
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 107, in main
    cli(None, None, None)
  File "c:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1024, in invoke
    Command.invoke(self, ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 76, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "c:\python27\lib\site-packages\platformio\maintenance.py", line 49, in on_platformio_start
    after_upgrade(ctx)
  File "c:\python27\lib\site-packages\platformio\maintenance.py", line 176, in after_upgrade
    ctx.invoke(cmd_platforms_update)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\commands\platforms.py", line 169, in platforms_update
    p.update()
  File "c:\python27\lib\site-packages\platformio\platforms\base.py", line 397, in update
    pm.update(name)
  File "c:\python27\lib\site-packages\platformio\pkgmanager.py", line 150, in update
    click.echo("[%s]" % (click.style("Up-to-date", fg="green")))
  File "c:\python27\lib\site-packages\click\utils.py", line 314, in echo
    file.write(message)
  File "c:\python27\lib\site-packages\click\_compat.py", line 560, in _safe_write
    return _write(s)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 166, in write_and_convert
    self.write_plain_text(text, cursor, start)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "c:\python27\lib\site-packages\click\_compat.py", line 97, in write
    return self.buffer.write(str(x))
IOError: [Errno 0] Error

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================

The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

**The content of `platformio.ini`:**
```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
upload_speed = 921600
board_f_cpu = 160000000L
```

Any help appreciated.

ivankravets, 2017-11-24:
Do you use the latest VSCode?

ivankravets, 2017-11-24:
if you still have an issue, please move it to https://github.com/platformio/platformio-vscode-ide/issues

closed 2017-11-24 by ivankravets

tekk, 2017-11-25:
Yes

tekk, 2017-11-25:
Reinstalled everything and now it works...

ivankravets, 2017-11-25:
Did you reinstall Python too?

tekk, 2017-11-25:
I'm not sure but most probably yes.

--------------------------
Ing. Peter Javorský

gusarov, 2019-01-07:
same here... I suspect that this might be related to the fact that initial pio installation been done while VS been UAC-Elevated...
I've tried to remove .platformio and start over again - still have issue "exit code 2". But it works from commandline

ivankravets, 2019-01-09:
@gusarov please file an issue here https://github.com/platformio/platformio-vscode-ide/issues